pagermon/pagermon #10: Support other DB types

davidmckenzie, 2017-06-06:
* MySQL
* Mongo?
* DynamoDB + Lambda?

davidf2002, 2018-02-24:
mysql/mariadb support would be great!

davidmckenzie, 2018-11-22:
This is neat, could come in handy for DB upgrade scripts on other DB types: https://itnext.io/updating-an-sql-database-schema-using-node-js-6c58173a455a

DanrwAU, 2018-11-22:
I've been playing around with this 
https://knexjs.org
Haven't got anywhere near being able to drop it in to replace the current stuff but it seems pretty neat. 
The only stumbling block will be the searching :(

DanrwAU, 2018-11-23:
@davidmckenzie i started playing with Knex while i was bored today. 

Only did the DB creation stuff in db.js - api.js gives me a headache.....
https://github.com/DanrwAU/pagermon/blob/knex/server/db.js

Thoughts?

davidmckenzie, 2018-11-23:
Ooh that looks pretty neat, and yeah, the API routes are gonna be the real herculean task lol. Will have to have a look at the knex docos at some point, see how it handles synchronicity. Does look super promising though

DanrwAU, 2019-04-26:
@davidmckenzie Happy to close this one for 0.3.0?

davidmckenzie, 2019-04-26:
Yep! I'd still like to do mongo support one day, but that'd be pretty much a complete fork so happy to close this. Thanks for all your work on it dude :)

PR #174 (2019-04-26): Support Multiple Databases
# Description

Complete re-write of database driver to remove native SQLite 3 client and replace with Knex query builder.

Add support for MySQL with possiblity of Amazon RDS and Postgres to be implemented later

Fixes #10 

## Type of change

- [x] Breaking change (fix or feature that would cause existing functionality to not work as expected)
- [x] This change requires a documentation update

# How Has This Been Tested?

**Need to Enter Tests Completed**

# Checklist:

- [x] My code follows the style guidelines of this project
- [x] I have performed a self-review o [... cut]
closed 2019-04-26 by davidmckenzie