pagermon/pagermon #24: Fontawesome Icon Picker

davidmckenzie, 2017-06-10:
Library I was planning on using is out of date, will need to research alternatives. For now, will place a link to the icon lists in the settings page.

https://github.com/itsjavi/fontawesome-iconpicker

shutterfire, 2019-01-05:
For some reason I can only use a very limited set of icons. Icons with a dash in their name seem to not work at all (heart-rate, briefcase-medical or something like that)

andrewerrington, 2022-03-20:
Came here looking for a solution, but I'll second the request to find and fix whatever it is that prevents fontawesome icons containing a dash in their name from working.

DanrwAU, 2022-03-20:
> Came here looking for a solution, but I'll second the request to find and fix whatever it is that prevents fontawesome icons containing a dash in their name from working.

Haven't encountered this and we aren't aware of it. Open an new issue please

eopo, 2024-04-15:
Closed by #514

closed 2024-04-15 by eopo