pagermon/pagermon #28: website

joper112, 2017-06-12:
Hi you just said :Now login via the website, default credentials are 'admin' / 'changeme'

But what is the website where did you stored the website data?

davidmckenzie, 2017-06-12:
Wherever you run 'pm2 start server/process.json' :)

Default port is 3000, so if you're running it on your local machine it would be http://localhost:3000/

joper112, 2017-06-12:
I run the command and installed everything but i cant reach localhost 3000

davidmckenzie, 2017-06-12:
Anything in the log files? Should be in server/logs.

Could you post the output of the command "pm2 show pagermon"?

joper112, 2017-06-13:
Hi my output is : 
┌───────────────────┬───────────────────────────────────────────────────┐
│ status            │ online                                            │
│ name              │ pagermon                                          │
│ restarts          │ 34                                                │
│ uptime            │ 0s                                                │
│ script path       │ /home/pi/pagermon/server/app.js                   │
│ script args       │ N/A                                               │
│ error log path    │ /home/pi/pagermon/server/logs/node-app.stderr.log │
│ out log path      │ /home/pi/pagermon/server/logs/node-app.stdout.log │
│ pid path          │ /home/pi/pagermon/server/pids/node-geo-api-1.pid  │
│ interpreter       │ node                                              │
│ interpreter args  │ --harmony                                         │
│ script id         │ 1                                                 │
│ exec cwd          │ /home/pi/pagermon/server                          │
│ exec mode         │ cluster_mode                                      │
│ node.js version   │ 7.10.0                                            │
│ watch & reload    │ ✘                                                 │
│ unstable restarts │ 7                                                 │
│ created at        │ 2017-06-13T07:09:29.407Z                          │
└───────────────────┴───────────────────────────────────────────────────┘
 Add your own code metrics: http://bit.ly/code-metrics
 Use `pm2 logs pagermon [--lines 1000]` to display logs
 Use `pm2 monit` to monitor CPU and Memory usage pagermon

davidmckenzie, 2017-06-13:
```
│ restarts │ 34 │
│ uptime │ 0s │
```

Looks like it's dying after start up - anything in the log files?

I see you're using node v7 too - I haven't tested with that version yet, but it *should* work. My servers are running on the stable branch - v6.11.0.

joper112, 2017-06-13:
Thats correct i restarted my server so i dit run it again. Can i downgrade my node? I just deleted everything and installed it again but now i get the error: 

sudo npm install
npm WARN saveError ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN pi No description
npm WARN pi No repository field.
npm WARN pi No README data
npm WARN pi No license field.

up to date in 0.93s

I dont have the file package.json.... 😢

davidmckenzie, 2017-06-13:
package.json should be in the server directory :)

joper112, 2017-06-13:
I installed it again but by sudo npm install

I got a few errors:
npm WARN saveError ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN pi No description
npm WARN pi No repository field.
npm WARN pi No README data
npm WARN pi No license field.

up to date in 0.93s
pi@raspberrypi:~ $ cd pagermon/
pi@raspberrypi:~/pagermon $ ls
client  LICENSE  README.md  server
pi@raspberrypi:~/pagermon $ cd server/
pi@raspberrypi:~/pagermon/server $ ls
app.js  config  package.json  process-default.json  public  routes  views
pi@raspberrypi:~/pagermon/server $ sudo npm install
npm WARN deprecated node-uuid@1.4.8: Use uuid module instead
npm WARN deprecated isarray@2.0.1: Just use Array.isArray directly

> uws@0.14.5 install /home/pi/pagermon/server/node_modules/uws
> node-gyp rebuild > build_log.txt 2>&1 || exit 0

> sqlite3@3.1.8 install /home/pi/pagermon/server/node_modules/sqlite3
> node-pre-gyp install --fallback-to-build

node-pre-gyp info it worked if it ends with ok
node-pre-gyp verb cli [ '/usr/bin/nodejs',
node-pre-gyp verb cli   '/home/pi/pagermon/server/node_modules/sqlite3/node_modules/.bin/node-pre-gyp',
node-pre-gyp verb cli   'install',
node-pre-gyp verb cli   '--fallback-to-build' ]
node-pre-gyp info using node-pre-gyp@0.6.31
node-pre-gyp info using node@7.10.0 | linux | arm
node-pre-gyp verb command install []
node-pre-gyp info check checked for "/home/pi/pagermon/server/node_modules/sqlite3/lib/binding/node-v51-linux-arm/node_sqlite3.node" (not found)
node-pre-gyp http GET https://mapbox-node-binary.s3.amazonaws.com/sqlite3/v3.1.8/node-v51-linux-arm.tar.gz
node-pre-gyp http 403 https://mapbox-node-binary.s3.amazonaws.com/sqlite3/v3.1.8/node-v51-linux-arm.tar.gz
node-pre-gyp ERR! Tried to download(403): https://mapbox-node-binary.s3.amazonaws.com/sqlite3/v3.1.8/node-v51-linux-arm.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for sqlite3@3.1.8 and node@7.10.0 (node-v51 ABI) (falling back to source compile with node-gyp)
node-pre-gyp http 403 status code downloading tarball https://mapbox-node-binary.s3.amazonaws.com/sqlite3/v3.1.8/node-v51-linux-arm.tar.gz
node-pre-gyp verb command build [ 'rebuild' ]
gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/7.10.0"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/pi/pagermon/server/node_modules/sqlite3/.node-gyp"
make: Map '/home/pi/pagermon/server/node_modules/sqlite3/build' wordt binnengegaan
make: *** Er is geen regel om doel '../.node-gyp/7.10.0/include/node/common.gypi' te maken, nodig voor 'Makefile'.  Gestopt.
make: Map '/home/pi/pagermon/server/node_modules/sqlite3/build' wordt verlaten
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:258:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:194:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Linux 4.9.29-v7+
gyp ERR! command "/usr/bin/nodejs" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/home/pi/pagermon/server/node_modules/sqlite3/lib/binding/node-v51-linux-arm/node_sqlite3.node" "--module_name=node_sqlite3" "--module_path=/home/pi/pagermon/server/node_modules/sqlite3/lib/binding/node-v51-linux-arm"
gyp ERR! cwd /home/pi/pagermon/server/node_modules/sqlite3
gyp ERR! node -v v7.10.0
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute '/usr/bin/nodejs /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/pi/pagermon/server/node_modules/sqlite3/lib/binding/node-v51-linux-arm/node_sqlite3.node --module_name=node_sqlite3 --module_path=/home/pi/pagermon/server/node_modules/sqlite3/lib/binding/node-v51-linux-arm' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/pi/pagermon/server/node_modules/sqlite3/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:106:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:194:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:899:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
node-pre-gyp ERR! System Linux 4.9.29-v7+
node-pre-gyp ERR! command "/usr/bin/nodejs" "/home/pi/pagermon/server/node_modules/sqlite3/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/pi/pagermon/server/node_modules/sqlite3
node-pre-gyp ERR! node -v v7.10.0
node-pre-gyp ERR! node-pre-gyp -v v0.6.31
node-pre-gyp ERR! not ok
Failed to execute '/usr/bin/nodejs /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/pi/pagermon/server/node_modules/sqlite3/lib/binding/node-v51-linux-arm/node_sqlite3.node --module_name=node_sqlite3 --module_path=/home/pi/pagermon/server/node_modules/sqlite3/lib/binding/node-v51-linux-arm' (1)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! sqlite3@3.1.8 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the sqlite3@3.1.8 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2017-06-13T07_43_22_440Z-debug.log

davidmckenzie, 2017-06-13:
```
node-pre-gyp http GET https://mapbox-node-binary.s3.amazonaws.com/sqlite3/v3.1.8/node-v51-linux-arm.tar.gz
node-pre-gyp http 403 https://mapbox-node-binary.s3.amazonaws.com/sqlite3/v3.1.8/node-v51-linux-arm.tar.gz
node-pre-gyp ERR! Tried to download(403): https://mapbox-node-binary.s3.amazonaws.com/sqlite3/v3.1.8/node-v51-linux-arm.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for sqlite3@3.1.8 and node@7.10.0 (node-v51 ABI) (falling back to source compile with node-gyp)
node-pre-gyp http 403 status code downloading tarball https://mapbox-node-binary.s3.amazonaws.com/sqlite3/v3.1.8/node-v51-linux-arm.tar.gz
```

So that looks like the [node-sqlite3](https://github.com/mapbox/node-sqlite3) plugin doesn't have binaries for ARM platforms. May be worth popping over to its repo if you can't get it built.

That said, running npm install as root is generally a bad idea, strange things can happen. You could maybe try running:

```
npm install node-sqlite3 --build-from-source
```

joper112, 2017-06-13:
meh A shit load of erros and:

gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:258:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:194:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Linux 4.9.29-v7+
gyp ERR! command "/usr/bin/nodejs" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/pagermon/node_modules/node-sqlite3
gyp ERR! node -v v7.10.0
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/pagermon/package.json'
npm WARN pagermon No description
npm WARN pagermon No repository field.
npm WARN pagermon No README data
npm WARN pagermon No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! node-sqlite3@0.0.3 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the node-sqlite3@0.0.3 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2017-06-13T08_21_41_154Z-debug.log

davidmckenzie, 2017-06-13:
I just ran a test now on a spare raspi I've got running a stream. Worked all good, though it did show some scary warnings on the compile from source for sqlite. This is running on Raspbian 8 (jessie), kernel 4.4.38-v7+.

Here's my bash history if it helps - note that I tested on node version 6 using nvm to switch versions:

```
  618  sudo apt-get update && sudo apt-get upgrade -y
  619  sudo apt-get install npm sqlite3
  621  curl -o- https://raw.githubusercontent.com/creationix/nvm/v0.33.2/install.sh | bash
  622  export NVM_DIR="$HOME/.nvm"
  623  [ -s "$NVM_DIR/nvm.sh" ] && \. "$NVM_DIR/nvm.sh" 
  624  [ -s "$NVM_DIR/bash_completion" ] && \. "$NVM_DIR/bash_completion"
  625  nvm install v6.11.0
  626  nvm use v6.11.0
  632  cd pagermon/
  633  cd server/
  635  cp process-default.json process.json
  636  vi process.json
  637  npm install npm@latest -g
  638  npm install pm2 -g
  639  npm install
  641  pm2 start process.json
```
No errors after the start, so confirmed all was working:

```
  642  pm2 show pagermon
  643  ls -la logs
  644  curl localhost:3000
  647  sqlite3 messages.db
  648  pm2 stop all
```

Curl returned the website as expected, and I could see the autocreated tables in sqlite3 with the command ".schema" :)

I can see quite a few bugs listed on the node-sqlite3 repo for raspi related things. Maybe try updating everything - there's also a suggestion to try running:

```
sudo apt-get install build-essential
npm install sqlite3 --unsafe-perm
```

Hope that helps!

joper112, 2017-06-13:
When i run pm2 start process.json and afther that pm2 show pagermon

status            │ errored                                           │
│ name              │ pagermon                                          │
│ restarts          │ 81                                                │
│ uptime            │ 0                                                 │
│ script path       │ /home/pi/pagermon/server/app.js                   │
│ script args       │ N/A                                               │
│ error log path    │ /home/pi/pagermon/server/logs/node-app.stderr.log │
│ out log path      │ /home/pi/pagermon/server/logs/node-app.stdout.log │
│ pid path          │ /home/pi/pagermon/server/pids/node-geo-api-1.pid  │
│ interpreter       │ node                                              │
│ interpreter args  │ --harmony                                         │
│ script id         │ 1                                                 │
│ exec cwd          │ /home/pi/pagermon/server                          │
│ exec mode         │ cluster_mode                                      │
│ node.js version   │ 7.10.0                                            │
│ watch & reload    │ ✘                                                 │
│ unstable restarts │ 0                                                 │
│ created at        │ N/A                                               │

He says errored do you know how i can change the node.js version? i used your bash from above but it doesnt work (no errors by installation)

joper112, 2017-06-13:
Failed to connect to localhost port 3000: Verbinding is geweigerd (translation connection reffussed.

davidmckenzie, 2017-06-13:
Maybe try with nodemon, should be a little more verbose:

```
pi@raspberrypi:~/pagermon/server $ npm install nodemon -g
/home/pi/.nvm/versions/node/v6.11.0/bin/nodemon -> /home/pi/.nvm/versions/node/v6.11.0/lib/node_modules/nodemon/bin/nodemon.js

> fsevents@1.1.2 install /home/pi/.nvm/versions/node/v6.11.0/lib/node_modules/nodemon/node_modules/fsevents
> node install

+ nodemon@1.11.0
added 256 packages in 215.957s
pi@raspberrypi:~/pagermon/server $ nodemon app.js
[nodemon] 1.11.0
[nodemon] to restart at any time, enter `rs`
[nodemon] watching: *.*
[nodemon] starting `node app.js`
```

joper112, 2017-06-14:
Thanks for your time to search for my problem. But when u run the command you say, the first part is no problem but when i run: nodemon app.js happends this:
 pi@raspberrypi:~/pagermon/server $ nodemon app.js
pi@raspberrypi:~/pagermon/server $

joper112, 2017-06-14:
When i run npm install -g nodemon i get 
nodemon app.js
[nodemon] 1.11.0
[nodemon] to restart at any time, enter `rs`
[nodemon] watching: *.*
[nodemon] starting `node app.js`
[nodemon] restarting due to changes...
[nodemon] starting `node app.js`

When i installed it by your command ( npm install nodemon -g) i cant run WEIRD! but i troubleshoot something and now its starting

joper112, 2017-06-14:
I fixed it. When i run pm2 start app.js its works

joper112, 2017-06-14:
Sorry new problem when i set my multimon and rtlfm config in the reader.sh and i runit i get

Error: Cannot find module 'nconf'
    at Function.Module._resolveFilename (module.js:470:15)
    at Function.Module._load (module.js:418:25)
    at Module.require (module.js:498:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/home/pi/pagermon/client/reader.js:25:13)
    at Module._compile (module.js:571:32)
    at Object.Module._extensions..js (module.js:580:10)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)

This is what i changed: 
rtl_fm -f 169.65M -s 22050 -p -19 |
multimon-ng -a FLEX -a SCOPE -t raw /dev/stdin |
node reader.js

davidmckenzie, 2017-06-14:
Yay! Glad the server bit's working now! :)

The client library also needs `npm install` to be run from its directory, so similar to server just cd in to pagermon/client and run `npm install`. The libraries aren't shared between them to keep it light-weight when the client is run on a separate machine.

joper112, 2017-06-14:
works! sorry......

i generated a api copied in the config but no output from reader.sh is in the database
is it possible to run the .sh without rtlfm and multimonng but only get the output from /dev/stdin? its because my multimonng is not the original for decoding flex now i got a lot of unk 00000000 messages

davidmckenzie, 2017-06-14:
Aha! I've been wondering what FLEX messages look like :)

So I've only had exposure to POCSAG and haven't done any regex for any other protocols in the reader.

What does the raw output from multimon look like, and how should it be cut up?

The POCSAG messages look like this:

    POCSAG512: Address: 1000000  Function: 0  Alpha:   Test message, disregard<EOT>

So I'm just using regex to grab the details:

    address = line.match(/POCSAG512: Address:(.*?)Function/)[1].trim();
    message = line.match(/Alpha:(.*?)$/)[1].trim();

Any ideas on the equivalent in the FLEX output? Guessing the figure between the square brackets is the address, and then all the content after that is the message?

jklnz, 2017-06-14:
This is what the messages look like:
FLEX - Always and only flex.
2017-06-14-00:03:26 time
1600/2/A baud rate/levels/phase
00.120 the flex cycle number . frame number (not needed)
[001234567] Address/capcode
ALN - message type / can be ALN - alpha numeric, TON for TONE or NUM for NUMERIC, or UNK for unknown
then the message is after that

FLEX: 2017-06-14 00:03:26 1600/2/A 00.120 [001234567] ALN THIS IS A TEST PERIODIC PAGE SEQUENTIAL NUMBER  9116

davidmckenzie, 2017-06-14:
Perfect, thanks!

Do you want to only store/display the ALN messages? I'm ignoring anything but alpha on POCSAG, not sure if you'd want the same?

jklnz, 2017-06-14:
I personally store the Alpha and Numeric ones (over here there is only like 2 numeric messages per day!) but I dont differentiate, I just treat them the same

jklnz, 2017-06-14:
Also there is a GPN message type too for group messages, i'm not sure how that comes up as they don't use group messages in NZ

joper112, 2017-06-14:
I need the ALN messages and de GPN groupmessages. The ALN are 1 ambulance or one capcode but by the firedepartment all of the messages are gpn because the alarmed more then one unit of a chief of someone for gas intoxication. blablabla 😄

davidmckenzie, 2017-06-14:
Awesome, think I've got some regex that'll cover it - just ran a few tests with the messages you guys posted:

```
user:~/pagermon/client$ echo "FLEX: 2017-06-14 00:03:26 1600/2/A 00.120 [001234567] ALN THIS IS A TEST PERIODIC PAGE SEQUENTIAL NUMBER 9116" | node reader.js
2017-06-14 20:11:01: 001234567: THIS IS A TEST PERIODIC PAGE SEQUENTIAL NUMBER 9116
Input died!
user:~/pagermon/client$ echo "FLEX: 2017-06-14 11:04:10 1600/2/A 01.015 [001520320] ALN A1 gaarne kontakt mka" | node reader.js
2017-06-14 20:11:20: 001520320: A1 gaarne kontakt mka
Input died!
user:~/pagermon/client$ echo "FLEX: 2017-06-14 11:03:44 1600/2/A 01.001 [002029568] GPN GLM_Prio 3 OMS BR (OMS/PAC oorzaak: las-/slijpwerkz.) #ELS-4406# O8234 PLURYN Jp Heije Jasappel 31 ELS" | node reader.js
2017-06-14 20:11:42: 002029568: GLM_Prio 3 OMS BR (OMS/PAC oorzaak: las-/slijpwerkz.) #ELS-4406# O8234 PLURYN Jp Heije Jasappel 31 ELS
Input died!
user:~/pagermon/client$ echo "FLEX: 2017-06-14 00:03:26 1600/2/A 00.120 [991234567] ALN THIS IS A TEST PERIODIC PAGE SEQUENTIAL NUMBER 9116" | node reader.js
2017-06-14 20:14:04: 991234567: THIS IS A TEST PERIODIC PAGE SEQUENTIAL NUMBER 9116
Input died!
```

They seem to have come through all good: https://hrng.io/

Only issue is with the length of the addresses - I've set the display to pad to 7 chars for a uniform display, it looks to be cutting off the first two digits of any addresses longer than that. Will have a look tomorrow at exposing a setting in the admin panel to control how many characters it should pad to :)

davidmckenzie, 2017-06-14:
Longer capcodes should now display properly - kept it simple, just replaced the existing padZeroes function with one that doesn't chomp addresses that are longer than 7 digits. May look at adding some configuration for how many chars to pad in future, but this should be good for 99% of use cases.

jklnz, 2017-06-14:
Yeah that looks good - every address i've seen on our networks here has been 7 or less characters.

joper112, 2017-06-15:
Maybe you can help. I run ./reader.sh in the client. The app is running and the pagermon site is visible only the are no messages. I just downloaded the fix but it doesnt work (i think)

davidmckenzie, 2017-06-15:
What does the console output look like @joper112? If it ignores a message then it'll show as grey - if it accepts it, it'll show the address in yellow and the message in blue.

Do the tests in the samples dir work?

joper112, 2017-06-16:
Everything in the output from reader.sh is (time red) (capcode/adress Yellow) (message blue) but nothing on localhost:3000 No messages found!
Have you loaded any message data yet?

davidmckenzie, 2017-06-17:
What do the logs say on the server side? You should be seeing some POSTs in the log when the message comes in. This is what a successful one looks like:

    2017-06-17 13:23 +00:00: nnn.nnn.nnn.nnn - - [17/Jun/2017:13:23:16 +0000] "POST /api/messages HTTP/1.1" 200 3 "-" "-"

If you're using PM2, the logs will be in server/logs/node-app.stdout.log

Can you grab about 10 lines either side of the POST in that log and add it here? :)

While you're in there, maybe check what the DB is showing with the following commands:

```
$ sqlite3 server/messages.db
sqlite> .schema
sqlite> SELECT * FROM messages LIMIT 5;
sqlite> SELECT * FROM capcodes LIMIT 5;
```

joper112, 2017-06-20:
@davidmckenzie the problem is there is no logs folder when i run the commands you say..... 

the only thing what i have is:

-rw-r--r--   1 pi pi 5,4K jun 15 12:55 app.js
drwxr-xr-x   2 pi pi 4,0K jun 15 13:34 config
-rw-r--r--   1 pi pi  12K jun 15 13:44 messages.db
drwxr-xr-x 142 pi pi 4,0K jun 15 13:10 node_modules
-rw-r--r--   1 pi pi 1,1K jun 15 13:10 package.json
-rw-r--r--   1 pi pi  51K jun 15 13:10 package-lock.json
-rw-r--r--   1 pi pi 1,1K jun 15 12:55 process-default.json
drwxr-xr-x   5 pi pi 4,0K jun 15 12:55 public
drwxr-xr-x   2 pi pi 4,0K jun 15 12:55 routes
-rw-r--r--   1 pi pi  12K jun 16 15:18 sessions.db
drwxr-xr-x   3 pi pi 4,0K jun 15 12:55 views

joper112, 2017-06-20:
The only thing wat is in my stdout is: 

cat node-app.stdout.log
2017-06-20 17:11 +02:00: ::1 - - [20/Jun/2017:15:11:52 +0000] "GET / HTTP/1.1" 200 24401 "-" "curl/7.38.0"
2017-06-20 17:13 +02:00: ::ffff:192.168.1.100 - - [20/Jun/2017:15:13:21 +0000] "GET / HTTP/1.1" 304 - "-" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_12_4) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/58.0.3029.110 Safari/537.36"
2017-06-20 17:13 +02:00: ::ffff:192.168.1.100 - - [20/Jun/2017:15:13:21 +0000] "GET /stylesheets/style.css HTTP/1.1" 304 - "http://192.168.1.18:3000/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_12_4) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/58.0.3029.110 Safari/537.36"
2017-06-20 17:13 +02:00: ::ffff:192.168.1.100 - - [20/Jun/2017:15:13:22 +0000] "GET /api/messages?limit=&page=1 HTTP/1.1" 304 - "http://192.168.1.18:3000/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_12_4) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/58.0.3029.110 Safari/537.36"
2017-06-20 17:13 +02:00: ::ffff:192.168.1.100 - - [20/Jun/2017:15:13:25 +0000] "GET /api/messages?limit=&page=1 HTTP/1.1" 304 - "http://192.168.1.18:3000/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_12_4) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/58.0.3029.110 Safari/537.36"
2017-06-20 17:13 +02:00: ::ffff:192.168.1.100 - - [20/Jun/2017:15:13:26 +0000] "GET /api/messages?limit=&page=1 HTTP/1.1" 304 - "http://192.168.1.18:3000/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_12_4) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/58.0.3029.110 Safari/537.36"
2017-06-20 17:13 +02:00: ::ffff:192.168.1.100 - - [20/Jun/2017:15:13:26 +0000] "GET /api/messages?limit=&page=1 HTTP/1.1" 304 - "http://192.168.1.18:3000/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_12_4) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/58.0.3029.110 Safari/537.36"
2017-06-20 17:14 +02:00: ::ffff:192.168.1.100 - - [20/Jun/2017:15:14:08 +0000] "GET /api/messages?limit=&page=1 HTTP/1.1" 304 - "http://192.168.1.18:3000/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_12_4) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/58.0.3029.110 Safari/537.36"

davidmckenzie, 2017-06-20:
Ok cool, thanks :)

I can't see any POSTs in the log, so looks like the client isn't hitting the server at the right address/port.

What's the content of your client/config.json (removing the API key)? Are you running the client on the same box as the server?

joper112, 2017-06-22:
My config file is : 

  "apikey": "MYAPIKEY ",
  "hostname": "http://127.0.0.1",
  "identifier": "p2000"
}

I run it on the same server i dont know what you mean by the same box? Maybe my indentifier is not good?

davidmckenzie, 2017-06-22:
Yup box = server :)

Looks like you're missing the port number in the hostname line - change that to "http://127.0.0.1:3000", see if that works.

Identifier line should be fine, it's just a plain text field, though keeping it to around 3 or 4 chars will improve how it looks in the table.

davidmckenzie, 2017-07-23:
How'd you end up going @joper112 ? Any issues still? :)

joper112, 2017-08-08:
@davidmckenzie it worked. Rasberry powered off now i nee to start everything again but the page is not coming online.  
netstat says only:
tcp6       0      0 :::3000                 :::*                    LISTEN      16044/app.js

but no page online

davidmckenzie, 2017-08-08:
If it's on port 16044 and you're not running a reverse proxy, you'd want your client settings to be http://<ip/hostname>:16044

closed 2018-02-25 by davidmckenzie