pagermon/pagermon #29: Regex matching

jklnz, 2017-06-12:
Alot of messages in NZ are in the same format (e.g. Ambo dispatch and Fire dispatch) so I use regex's to show data on them, e.g. fire messages all look like: 	

(PUKE431) 111-MIN 123 FAKE ST SAINT ANDREWS HAMILTON CITY. (XStr HEATH ST/BEERESCOURT RD) .ALARM ACTIVATION - RESIDENTIAL. #F1234567, 

So I have a webpage that uses the regex info to for instance translate what "PUKE431" means etc.

Are you planning on implementing a similar feature?

davidmckenzie, 2017-06-12:
Hmm, could be a really interesting feature. Can think of a few neat ways to do this... do you have an example of some of the regex you're using there? What would PUKE431 translate into?

jklnz, 2017-06-12:
In that example PUKE431 is the appliance so usually i put a thing in so you can hover over it and it says what ever the appliances full name is (e.g. Pukete 431).

This is the current Regex (please note it is in .net format) I use for NZ Fire Service:
\((?<FireAppliances>.+?)\) (?<FireCallsource>.+?)-(?<FireCalltype>.+?)( \(Alarm Type (?<FireAlarmType>.+?)\))?( \(Box (?<FireBox>.+?)\))? (?<Address>.+?). \(XStr (?<XStreet1>.*?)/(?<XStreet2>.*?)\)(?<k> \.)?(?<FireDetails>.+?\.)? (?<MapGrid>\(x-\d{1,10} y-\d{1,10}\))?(?<Additional>.*)? ?#(?<FireEventNumber>F\d+)

Then i have a lookup list of all the values for each of the relevant field types. E.g. for FireCalltype i have a list item that says "STRU" is "Structure Fire" etc.

[image]

davidmckenzie, 2017-06-12:
Cool, makes sense. Could expand the [angular-highlight](https://github.com/26medias/angular-highlight) module to do something like that. It works on an array at the moment, applying HTML around keywords matching that array. E.g. the following is generated from having "ALERT" and "FIRECALL" as highlight keywords:

```
<div highlight="message.message" keywords="init.highlightText" class="pagerMessage ng-isolate-scope">
<a href="/?q=ALERT" class="angular-highlight">ALERT</a> SpriHi - <a href="/?q=FIRECALL" class="angular-highlight">FIRECALL</a> - #101722 - CARCOAR ST SPRING HILL
</div>
```

If we make it work with an array of objects you could have a keyword and a bit of "translation text" that shows up as a tooltip or something.

jklnz, 2017-06-13:
Sweet, that sounds good, I really want to switch to your system at some stage soon, as my one is getting a bit old and your one looks like it would be a good replacement.

jklnz, 2017-06-13:
One other suggestion is that over here we have two pager networks so maybe adding a network to the pagermessages table?

davidmckenzie, 2017-06-13:
Easy :) Will call it a 'source' field, so it's flexible as either a name for the client machine/location or a network, etc.

Gotta do some DB restructuring in #30 anyway, so shouldn't be too big an issue.

davidmckenzie, 2017-06-14:
Getting close to something workable for the regex matching - in your opinion would it be better to have the expanded text as a tooltip similar to how you have it now? Or would it be better to replace the text (maybe with original text in a tooltip)?

Demo: https://pagermon-hrng.c9users.io:8080/

So on that page I'm replacing "test" with "PUPPY", "fire" with "PARTY", and "[0-9][0-9][0-9][0-9]" with "numbers".

davidmckenzie, 2017-06-14:
Also, with the source field - I'm thinking keep it to a real short identifier for optimal UX. Something like 4 or 5 chars max, would that work for you? Would be a column next to the capcode field, hidden on mobile.

jklnz, 2017-06-14:
re the Source - Yeah that would be great - currently I use really simple IDs for the network: Amb for Ambulance and TP2 for the main network.

jklnz, 2017-06-14:
re the Regex - i think putting tool tips would probably be the best as most of the time you know what it is but the time you do need to know what it means you can just hover over it...

davidmckenzie, 2017-06-14:
Too easy, both sound good to me. :)

davidmckenzie, 2017-06-14:
Ok cool, just testing the regex stuff but looks good so far:

https://pagermon-hrng.c9users.io:8080/

Settings:

http://i.imgur.com/D1GpJzO.jpeg

jklnz, 2017-06-14:
Sweet, that looks great! I will spin up a VM and test it out.

davidmckenzie, 2017-06-14:
Awesome :) Have just merged the PR, so should be good to go.

Feel free to reopen if you run into any trouble!

closed 2017-06-14 by davidmckenzie