pagermon/pagermon #33: Error on 0.1.1 Admin pages

jklnz, 2017-06-14:
Hey I have set up PagerMon in my VM and whenever I browse to an admin page it says:
jquery.min.js:2 Uncaught Error: [$injector:modulerr] http://errors.angularjs.org/1.6.4/$injector/modulerr?p0=app&p1=Error%3A%20%…ogleapis.com%2Fajax%2Flibs%2Fangularjs%2F1.6.4%2Fangular.min.js%3A22%3A179)
    at angular.js:38
    at angular.js:4920
    at q (angular.js:403)
    at g (angular.js:4880)
    at eb (angular.js:4802)
    at c (angular.js:1914)
    at Sc (angular.js:1935)
    at ue (angular.js:1820)
    at HTMLDocument.<anonymous> (angular.js:33367)
    at j (jquery.min.js:2)

davidmckenzie, 2017-06-14:
That's weird - I got the same error once, but then subsequent visits worked fine, and now I can't replicate it 😕 

Could you change line 21 in /server/views/global/header.ejs from:

    <script src="//ajax.googleapis.com/ajax/libs/jquery/3.2.1/jquery.min.js"></script>

to:

    <script src="//ajax.googleapis.com/ajax/libs/jquery/3.2.1/jquery.js"></script>

And send through the error then? Should give a more verbose exception.

jklnz, 2017-06-14:
Try this log:
I changed the Angular from min to full to get it to work (jquery alone didnt change the logs)

[localhost-1497430681936.txt](https://github.com/davidmckenzie/pagermon/files/1074056/localhost-1497430681936.txt)

jklnz, 2017-06-14:
I tried refreshing a few times as well, and changing browser but still no joy!

davidmckenzie, 2017-06-14:
Whoops, yeah, angular, not jquery! 😅 

I think I see the issue. I've included angular-uuid with `//unpkg.com/angular-uuid@0.0.4/angular-uuid.js`. My theory is that's failing when you visit the site on http - I'm proxying all my traffic via https, so wouldn't have noticed it.

Gimme a couple of mins and I'll push a fix :)

davidmckenzie, 2017-06-14:
Ok, pushed! That should *hopefully* fix it - if not let me know if the same or different error. Might give a few browser warnings due to the cross-protocol includes.

jklnz, 2017-06-14:
Haha all good! 

No worries - I haven't set up anything advanced like HTTPS yet, just testing it out to see how to set it up etc.

jklnz, 2017-06-14:
Sweet, thanks i'll download it now

jklnz, 2017-06-14:
Hmmm, now I get:
Failed to instantiate module color.picker due to:
Error: [$injector:nomod] http://errors.angularjs.org/1.6.4/$injector/nomod?p0=colo...
    at http://ajax.googleapis.com/ajax/libs/angularjs/1.6.4/angular.min.js:6:425
    at http://ajax.googleapis.com/ajax/libs/angularjs/1.6.4/angular.min.js:26:270
    at b (http://ajax.googleapis.com/ajax/libs/angularjs/1.6.4/angular.min.js:25:299)
    at http://ajax.googleapis.com/ajax/libs/angularjs/1.6.4/angular.min.js:26:44
    at http://ajax.googleapis.com/ajax/libs/angularjs/1.6.4/angular.min.js:42:117
    at q (http://ajax.googleapis.com/ajax/libs/angularjs/1.6.4/angular.min.js:7:495)
    at g (http://ajax.googleapis.com/ajax/libs/angularjs/1.6.4/angular.min.js:41:476)
    at http://ajax.googleapis.com/ajax/libs/angularjs/1.6.4/angular.min.js:42:149
    at q (http://ajax.googleapis.com/ajax/libs/angularjs/1.6.4/angular.min.js:7:495)
    at g (http://ajax.googleapis.com/ajax/libs/angularjs/1.6.4/angular.min.js:41:476

davidmckenzie, 2017-06-14:
Pushed another quick change - that would be due to loading the modules in async - switched back to sync now.

jklnz, 2017-06-14:
Sweet as, that did it. Thanks!
Now that i've got it to work in my Windows VM I will start sending some test messages to it tomorrow.

closed 2017-06-15 by jklnz