pagermon/pagermon #34: Message field

jklnz, 2017-06-15:
One other question, is it possible to have another field in the, messages table to do the following:
basically on import i'm going to check the message against a set pattern/regex and if it matches then i'd like to record which pattern is used to match it - that way in the UI at some stage you would be able to filter it to only show messages which matched pattern x. 

This way when messages come in - like ambo dispatches - they can be seen and any other messages sent to the pager number are just in the background / only show up when you view by capcode etc.

Does that make sense? Let me know if not?

davidmckenzie, 2017-06-15:
Possibly! Have you got some examples so I can visualise it?

jklnz, 2017-06-16:
Yeah, it's a little hard to explain but essentially each pager message is checked on arrival to the server against the list of known regex patterns (probably best to have a table, at the moment I have five patterns)
One example for Ambulance jobs is
`(?<AmbulanceCallsign>\w{1,10}?)( )?(?<AmbulancePriority>AIR TRANSFER|GRAY|GREEN|GREY|NOTIFI|NOTIFICATION|ON HOLD|ORANGE|PRIVATE HIRE|PTS|PURPLE|RED|Test|TRIAGE) (?<AmbulanceCallType>\w+?) (?<AmbulanceCalltypeDescription>.+)?; Flat/Unit:(?<Flat>.+)?/(?<Address>.+)`

So when one of the messages come through that looks like:
`CHR33 RED 31D03 UNCON/FAINT NOT ALERT ; Flat/Unit: /123 Fake St BELFAST`

The system runs through all the patterns and checks if it matches, when it does match the ambulance dispatch one it adds whatever the name of the pattern is to the message row. That way in the UI you can view only the Ambulance dispatches.

jklnz, 2017-06-16:
Happy to send you an example picture of what I currently do off list if you want

davidmckenzie, 2017-06-16:
Cool, I get ya. Will have to have a think about that one, see what I can implement that's reusable in other scenarios too :)