pagermon/pagermon #36: Sending data to API

jklnz, 2017-06-17:
Hey @davidmckenzie , sorry for the million bugs/questions!
I'm sending this message to the server
`POCSAG512: 2017-06-17 13:54:30 [000780001] ALN HAPPY TO SAY YOU'VE BROKEN`
Using a custom program (that basically waits a minute for fragmented messages then combines them and sends them to the server).
If i send this message without the ' character it goes through OK, but if I keep it it will fail.
I tried URL encoding it but when I do that it does not decode them. Is there anything I should be doing to get the ' to come through properly?

jklnz, 2017-06-17:
Just FYI on my version of multimon - i changed the pocsag output to be in the same format as FLEX so that's why it looks the same

jklnz, 2017-06-17:
Without putting it on the debugger, my guess is this line (478/479) here in api.js:

            var dupeCheck = "SELECT * FROM messages WHERE id IN ( SELECT id FROM messages ORDER BY id DESC LIMIT "+dupeLimit;
                dupeCheck += " ) AND message LIKE '"+message+"' AND address="+address;

davidmckenzie, 2017-06-17:
Ooh nice catch. Node_sqlite3 has some built in SQLi protections, must be triggering something there. Will check it out, thanks!

jklnz, 2017-06-18:
Yeah it must be something like that. So far so good, it is working perfectly other than that. And it's really fast at showing messages too :)

davidmckenzie, 2017-06-18:
Should be all good now - converted to using double quotes inside the string, and stripped all double quotes from the message. Also fixed the error handling so it would actually throw an error if this somehow still happens 😄

closed 2017-06-18 by davidmckenzie