pagermon/pagermon #41: Messages not sorted by time in list?

jklnz, 2017-06-18:
[image]

Should messages be sorted by time in the message list?

davidmckenzie, 2017-06-18:
So it should be sorted by time... in theory. It's sorted by the message ID column, which is a unique key added on insert.

The datetime figure is set by the client component and is just a simple unix time. So I'm guessing something queued in the HTTP side causing the message to be processed out of order. Would be curious to see your log files on the server from around that time.

Other possibility is that the socket emit was out of order, but that would mean they would be in the right order after a refresh, which I'm guessing didn't happen? :)

jklnz, 2017-06-18:
That would explain it then, as the message to 0999182  at 1639 was fragmented and missing it's final piece so was held for one minute before uploading to PagerMon. So it came in after the two later ones at 1640. So the display is sorted by the Message ID which is why it looks like that.

davidmckenzie, 2017-06-18:
Ah! :)

So we could sort by the timestamp instead, but this might lead to inconsistent behaviour when multiple messages are broadcast during the same second. Alternatively we could move the timestamping to the server side, so it sets the time when it actually receives the message. Orrr we could ignore it and pretend it never happened. ;)

jklnz, 2017-06-18:
Haha Yeah I think it's best just to ignore it and pretend it never happened. Probably not a huge issue really.

davidmckenzie, 2017-06-18:
Will ignore with haste, thanks dude!

closed 2017-06-18 by davidmckenzie