pagermon/pagermon #42: Website feedback

jklnz, 2017-06-21:
Just some feedback for you mate, I have imported like 1000 capcodes from my other system and it now runs **much** slower than before when loading the index page and search pages.

I imagine it might be to do with the capcode - alias matching you have set up. Not a major issue really, but something to look at down the track?

davidmckenzie, 2017-06-21:
Definitely... I've only got a touch over 100 capcodes in mine, haven't seen any performance issues yet. One of the things I'll likely be prioritising sometime soon is getting it working with AWS's DynamoDB (particularly since it's free ;)) and locally run MongoDB. SQLite just doesn't scale well for this kinda thing.

Might see if we can get a quick win boost in performance by adding an index on the address column.

jklnz, 2017-06-21:
So far I have 927 capcodes and 14000 messages (over the past five days). 
I'm not 100% on SQLite (I tend to use MS SQL) but I did add an index on both tables and did not notice any improvement, hopefully you have better luck than me!
It is still really quick displaying the new pagers as they come in though.
Yeah either of those database options would be great!

closed 2017-06-29 by davidmckenzie