pagermon/pagermon #43: Websocket issue?

marshyonline, 2017-06-21:
Brower looks like it is trying to update, new line appears as follows
`Invalid date	Invalid date		undefined		Unknown	`

I do believe a message is coming at the time it does this, but the messages is set to reject.

Console Shows
> New Message ID: 167 currentPage: 1
angular.js:14525 TypeError: Cannot read property 'replace' of undefined
    at (index):40
    at m.$digest (angular.js:17999)
    at m.$apply (angular.js:18269)
    at l (angular.js:12387)
    at XMLHttpRequest.A.onload (angular.js:12541)
(anonymous) @ angular.js:14525
(anonymous) @ angular.js:11008
$digest @ angular.js:18017
$apply @ angular.js:18269
l @ angular.js:12387
A.onload @ angular.js:12541

jklnz, 2017-06-21:
Interesting, I think I am getting the same error here too. It was working fine until this morning when I got up and now it is constantly waiting on this network request:
http://192.168.1.143:3000/api/messages?limit=&page=1

jklnz, 2017-06-21:
This is the error message:
```
2017-06-22 08:29 +12:00: { Error: SQLITE_MISUSE: Database is closed
    at Database.<anonymous> (C:\Temp\pagermon-0.1.1-beta\pagermon-0.1.1-beta\server\node_modules\sqlite3\lib\sqlite3.js:20:25)
    at Database.object.(anonymous function) [as all] (C:\Temp\pagermon-0.1.1-beta\pagermon-0.1.1-beta\server\node_modules\sqlite3\lib\trace.js:31:20)
    at C:\Temp\pagermon-0.1.1-beta\pagermon-0.1.1-beta\server\routes\api.js:130:8
    at Layer.handle [as handle_request] (C:\Temp\pagermon-0.1.1-beta\pagermon-0.1.1-beta\server\node_modules\express\lib\router\layer.js:95:5)
    at next (C:\Temp\pagermon-0.1.1-beta\pagermon-0.1.1-beta\server\node_modules\express\lib\router\route.js:137:13)
    at Route.dispatch (C:\Temp\pagermon-0.1.1-beta\pagermon-0.1.1-beta\server\node_modules\express\lib\router\route.js:112:3)
    at Layer.handle [as handle_request] (C:\Temp\pagermon-0.1.1-beta\pagermon-0.1.1-beta\server\node_modules\express\lib\router\layer.js:95:5)
    at C:\Temp\pagermon-0.1.1-beta\pagermon-0.1.1-beta\server\node_modules\express\lib\router\index.js:281:22
    at Function.process_params (C:\Temp\pagermon-0.1.1-beta\pagermon-0.1.1-beta\server\node_modules\express\lib\router\index.js:335:12)
    at next (C:\Temp\pagermon-0.1.1-beta\pagermon-0.1.1-beta\server\node_modules\express\lib\router\index.js:275:10) errno: 21, code: 'SQLITE_MISUSE' }
```

jklnz, 2017-06-21:
Must have been an OS issue, restarted Windows and it works

davidmckenzie, 2017-06-22:
The DB closed error might have been due to a close statement I left in for error handling. Removed most of them but looks like a couple stayed in there. Have updated now, hopefully that prevents it happening again. :)

@marshyonline do the messages show up fine after reloading? Any errors in the stdout or stderr logs on the server?

marshyonline, 2017-06-22:
Messages show fine after a refresh, will check the logs tonight and report
back.

Will update, restart and test to confirm.

jklnz, 2017-06-22:
Sweet, I have updated and had no issues!

jklnz, 2017-06-23:
I think I know what caused this, I just created an Ignore filter on mine, and as soon as I did that some messages are coming through as you describe in the top comment.
I just watched another message come in and do exactly the same thing when it matched the ignore filter

jklnz, 2017-06-23:
Deleted the ignore filter and it works.

davidmckenzie, 2017-06-23:
Ah, good catch! Sounds like I need to add some better error handling on the ignored messages. Am offline tonight but will see if I can push a patch tomorrow :)

marshyonline, 2017-06-23:
Can confirm this has not solved the issue
http://imgur.com/a/piLis

I have a few ignore aliases, these messages hit the server roughly every minute, around the same time these error's are thrown.
My best guess is it is an issue with this somehow.

davidmckenzie, 2017-06-24:
Latest commit should resolve this - I ended up resolving #8 at the same time, so now the socket emits the full message. The client side still does a quick check if any filters are present, but if a message is set to be ignored (either through an alias or through enabling PDW mode), it will never be sent to the client.

Could you test and let me know how that goes? :)

marshyonline, 2017-06-24:
This has solved the issue, but in the process created another issue.
Loading Messages or changing pages is now pinning the CPU to max. 
Initial load time has increased dramatically, and the same with changing pages.
DB Currently is storing 14877 Messages with 737 capcodes

http://imgur.com/a/gi1GK

marshyonline, 2017-06-24:
This could also be due to #42 #

davidmckenzie, 2017-06-24:
Damn :( CPU just on server, or any browser performance issues too?

marshyonline, 2017-06-24:
Browser just seems to be waiting on a reply from the server. Not sure how I
can profile it when the server is taking it time to respond.

davidmckenzie, 2017-06-24:
No probs mate. I think I know what would be causing it anyway. Any chance I could get a copy of your messages.db file for testing? Can shoot you an email :)

marshyonline, 2017-06-24:
No worries, shoot me an email.
Address should be in my git profile

davidmckenzie, 2017-06-24:
Just a quick update for you guys - I dropped some timers in the api code to see what operations are taking the longest:

```
init: 1ms
sql: 4051ms
array: 4ms
send: 4ms
xx.xx.xx.xx - - [24/Jun/2017:22:06:06 +0000] "GET /api/messages?limit=100&page=1 HTTP/1.1" 200 - "https://pagermon-hrng.c9users.io/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_11_6) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/58.0.3029.110 Safari/537.36"
```

That SQL code is this bit:

```
    console.time('sql');
    var sql = "SELECT messages.*, capcodes.alias, capcodes.agency, capcodes.icon, capcodes.color, capcodes.ignore, MAX(capcodes.address) ";
        sql += " FROM messages";
        sql += " LEFT JOIN capcodes ON messages.address LIKE (capcodes.address || '%')";
        sql += " GROUP BY messages.id ORDER BY messages.id DESC";
    db.all(sql,function(err,rows){
        if (err) {
            console.log(err);
        } else if (rows) {
            console.timeEnd('sql');
```

Will try now with indexes :)

jklnz, 2017-06-25:
Just been profiling that statement in Sqlitestudio:
11.69 seconds: SELECT messages.*, capcodes.alias, capcodes.agency, capcodes.icon, capcodes.color, capcodes.ignore, MAX(capcodes.address) 
 FROM messages
LEFT JOIN capcodes ON messages.address LIKE (capcodes.address || '%') GROUP BY messages.id ORDER BY messages.id DESC
0.051seconds: SELECT messages.*, capcodes.alias, capcodes.agency, capcodes.icon, capcodes.color, capcodes.ignore, MAX(capcodes.address) 
 FROM messages
LEFT JOIN capcodes ON messages.address = capcodes.address GROUP BY messages.id ORDER BY messages.id DESC

Obviously that would wreck the wildcard matching (I don't use that, presume you do?)

jklnz, 2017-06-25:
I changed the relevant SQL statements on mine and it improves it massively,this is just a stop gap though until you find another option.

davidmckenzie, 2017-06-25:
Yeah it's definitely the wildcard matching that's killing it. Indexes don't do a thing when globbing like that. Have been able to scrape some marginal performance gains, but it's like two steps forward one step back at the moment.

Will mull it over for a while - I'm sure there's a smarter way to do this 😄

The array manipulation is super fast, so might look at just grabbing all messages and all capcodes into an array, and doing the joins on the javascript side. Not sure how that will scale when we start talking about millions of rows though.

jklnz, 2017-06-25:
Yeah I have been running a few options through SQLIte but no joy yet.

One way to keep the current DB system could be to specify the wildcard in the filter instead of presuming all filters are wildcard? This way in the query the only wildcards would be the ones you specified... Not sure if that would help you though depending on how many you have in your filters...

davidmckenzie, 2017-06-25:
Omg I'm an idiot :rofl: 

If the capcodes table address field is stored as text, using underscore as a single character wildcard, it looks like it works perfectly. That's what I had in mind originally when I was looking at doing the import from PDW's filters.ini. I don't know why, but I always assumed it would escape the special chars.

Will test out performance with that on @marshyonline's dataset, and I think I can rig up some kind of upgrade script for users that use the wildcard functionality, with an option to clear out the capcodes table and start fresh if you haven't modified since the filter import.

Will also have a think about what to do with leading 0s on this.

jklnz, 2017-06-25:
Nice, that sounds like it will work out well

davidmckenzie, 2017-06-25:
Ok, so running some tests, I think I have something workable. I'll be converting the client side scripts to send the full 7 char capcode for POCSAG (including leading zeroes), and a minimum of 7 chars for FLEX. Since from what I've seen, there's near 0 usage of > 7 char FLEX capcodes.

Both tables will have the address column changed to text, with some indexes added for good measure. Then will run the following to convert existing data:

```
UPDATE messages SET address = '0' || address WHERE LENGTH(address)=6;
UPDATE messages SET address = '00' || address WHERE LENGTH(address)=5;
UPDATE messages SET address = '000' || address WHERE LENGTH(address)=4;
UPDATE messages SET address = '0000' || address WHERE LENGTH(address)=3;
UPDATE messages SET address = '00000' || address WHERE LENGTH(address)=2;
UPDATE messages SET address = '000000' || address WHERE LENGTH(address)=1;
```
This will pad zeroes on the existing messages, so everything is neat and tidy.

```
UPDATE capcodes SET address = '0' || address WHERE LENGTH(address)=6;
UPDATE capcodes SET address = '%' || address || '%' WHERE LENGTH(address)<6;
```
For 6 char entries in the capcodes table, I'm assuming they're not wildcard based and popping a zero in front. For anything less than 6 chars, I'll assume they are wildcard based and will wrap in % so that they match properly.

This would probably need some manual clean-up after, depending on your data. I know I've got a few 2 or 3 char aliases that I'll need to update.

Did some testing with this setup, and it goes from >4000ms per query to ~700ms per query. Got a few more ideas to milk a bit more performance out of it though.

jklnz, 2017-06-25:
Sweet, that's sounding great.

Looks like that checks in with my capcode database too. None of mine are under 6 chars long, 848 are 6 char long and 150 are 7 char long.
I had a few shorter capcode aliases but they were causing too many false positives so I will just re-add them afterwards.
My message DB currently has 25k messages in it.

davidmckenzie, 2017-06-25:
Do you use the "PDW mode" where messages without an alias are ignored?

jklnz, 2017-06-25:
Nah I show them all

davidmckenzie, 2017-06-26:
Getting closer now :)

Follow up question for you guys - would you prefer if aliases that are set to ignore/filter out were not saved to the DB?

Would mean that you couldn't untick the ignore button and have them magically come back, but would also mean a good boost to performance not having a whole bunch of stale messages in the DB.

I'm leaning towards dropping them, but depends on how you guys are using the system.

marshyonline, 2017-06-26:
The way I see it, I'd you have set an ignore filter it's garbage anyway, no point storing anything that has an alias set to ignore.
Aslong as that does not affect messages that have no Alias then I don't see an issue with it.

jklnz, 2017-06-26:
Yeah I would have to agree with @marshyonline - I only have two ignore filters set and they are both for messages which I probably don't want to see again so should be good to just bin them.

davidmckenzie, 2017-06-26:
Perfect, thanks :D

davidmckenzie, 2017-06-26:
Response times getting lower now - getting sub 200ms query time for @marshyonline's data.

Helps when like 90% of it matched the ignore aliases though.

Should be able to get some further optimisations - since the ignored data no longer needs to be filtered out, we can do some pagination in the DB side, which means no more parsing gigantic arrays full of misery and tears.

@jklnz - any chance I could get a copy of your messages.db file too for testing?

jklnz, 2017-06-26:
Sweet, that'll be good!

Yeah what's your email?

davidmckenzie, 2017-06-26:
@jklnz dm@hrng.io :)

Updated times - with 'PDW Mode' enabled:

```
init: 178ms
sql: 8ms
send: 3ms
GET /api/messages?limit=20&page=1
```

Without PDW mode (so all messages):

```
init: 1ms
sql: 1ms
send: 2ms
GET /api/messages?limit=20&page=1
```

The increased time on PDW mode is due to a join required to get the count of messages, should be able to reduce that further though.

```
    if (pdwMode) {
        initSql =  "SELECT COUNT(*) AS msgcount FROM messages";
        initSql += " INNER JOIN capcodes ON capcodes.id = (SELECT id FROM capcodes WHERE address LIKE messages.address LIMIT 1);";
    } else {
        initSql = "SELECT COUNT(*) AS msgcount FROM messages;";
    }
```

davidmckenzie, 2017-06-26:
Searching is still a little expensive, will need to see how this goes with a bigger dataset:

PDW mode (311 messages)
```
init: 0ms
sql: 201ms
search: 60ms
GET /api/messageSearch?limit=20&page=1&q=FIRE
init: 0ms
sql: 184ms
search: 42ms
GET /api/messageSearch?limit=20&page=7&q=FIRE
```

All messages (2619 messages)
```
init: 0ms
sql: 333ms
search: 281ms
GET /api/messageSearch?limit=20&page=1&q=FIRE
init: 0ms
sql: 208ms
search: 197ms
GET /api/messageSearch?limit=20&page=7&q=FIRE
```

Can't paginate the results in SQL since we need to do full text searching across multiple columns - ends up quicker to just grab all the data and parse the big-ass array.

I've been thinking about ways to do smarter filtering anyway, so now might be a good time to implement it. What I'd really like to be able to do is right click a capcode/agency/source/alias and say "Filter out" or "Show matching" - you could do this multiple times to construct a nice bookmarkable link that shows only the data you want.

Would still keep the full text search for message content, but have all other columns filtered smarter.

davidmckenzie, 2017-06-26:
Testing @jklnz's dataset now, PDW mode slows to a crawl - looks like that option will always be impacted by the number of messages:

PDW mode (14288 messages)
```
init: 1469ms
sql: 4ms
send: 2ms
```

All messages (27295 messages)
```
init: 1ms
sql: 2ms
send: 1ms
```

So I guess some good news and bad news :)

Full text search struggles even more:
```
init: 0ms
sql: 1785ms
search: 2530ms
GET /api/messageSearch?limit=20&page=1&q=acute
init: 0ms
sql: 1773ms
search: 4719ms
GET /api/messageSearch?limit=20&page=10&q=acute
```

jklnz, 2017-06-26:
Glad I don't use PDW mode hahaha
I do use the search a bit though.

davidmckenzie, 2017-06-27:
Quick update - haven't been able to find a neat way around the speed of PDW mode with large record sets, so I'm going to ignore that for now and concentrate on optimising the search functionality. Once that's looking better, I'll push this code out and then prioritise #10 - as that will always be the better choice for the PDW mode use cases :)

davidmckenzie, 2017-06-28:
Idea time!

Add a foreign key column to the messages table, which links to a corresponding alias ID. This field is updated either by pressing a button in settings after making changes to capcode aliases, or as a trigger on the capcodes table (former would be better, especially for bulk updates). This would make both things super fast, as you could get the count simply by whether or not that foreign key is populated.

Will have a look at that tonight, see whether it's plausible.

jklnz, 2017-06-28:
I think that's probably the best idea. I think that will improve performance a lot, as you would only need to do that wild card search once instead of 10,000x when you do the query.
Couldn't you just do it on the insertion code, just check if the alias matches then add the FK if it does? No need for a trigger?

davidmckenzie, 2017-06-28:
Yeah will see what performance is like - it might be a bit much to have a 5-10 second delay whenever you update an alias, since it would have to go through all of them every time (i.e. you wouldn't want to create an alias for '%' and have it overwrite the alias for everything else).

jklnz, 2017-06-28:
Maybe have an option saying do you want to update existing messages?

davidmckenzie, 2017-06-28:
Looking a lot better now, refactored searching to use proper queries. The filters on agency/capcode/source are between 100ms and 500ms depending on which page you're on and what you're searching (being on the last page of a filter on the source had the longest result, filtering on capcode was the quickest).

The full text search still sucks, but is bearable. Takes about 2s on average for a really broad search term. Further improvements there will have to wait until either #10 or when I construct a more advanced filtering page.

I'm thinking with the alias updates we'll have two buttons - "Save" and "Save & Update" with help text to indicate the latter will update existing messages, and may take a while on large databases. Will also add a button next on the alias list page to trigger an update. It'll eventually become a problem again when message databases keep growing, but that's a problem for future Dave (probably solved by #12 and #10 anyway).

In other news - on the front page view (so no filters/searches), both PDW mode and non-PDW mode are logging <10ms processing times from first page to last. Yay!

marshyonline, 2017-06-28:
This makes me very happy!
Great work, and thank you!

jklnz, 2017-06-28:
Nice work. Regarding #12 - can you make rotation option?, as in my old system I keep all messages  so ideally want to do the same... currently it has 1.4million messages.

davidmckenzie, 2017-06-28:
Definitely will be optional :) We'll have to see how performance looks with a dataset that large after #10 - I imagine it wouldn't be much of an issue with a proper relational DB.

davidmckenzie, 2017-06-28:
Hrm, a potential issue with the wildcarding...

```
sqlite> SELECT * FROM capcodes WHERE '0401200' LIKE address ORDER BY address DESC;
id|address|alias|agency|icon|color|ignore
1118|0401___|test|RFS|question|black|0
121|0401200|Ruatoria|Ambo|medkit|#003300|0
1119|0401%|test wildcard|RFS|question|#000000|0
```
```
sqlite> SELECT * FROM capcodes WHERE '0401200' GLOB address ORDER BY address DESC;
id|address|alias|agency|icon|color|ignore
1118|0401???|test|RFS|question|black|0
121|0401200|Ruatoria|Ambo|medkit|#003300|0
1119|0401*|test wildcard|RFS|question|#000000|0
```
Both GLOB and LIKE are ordering wrong, may have to look at some fancy case ordering.

davidmckenzie, 2017-06-29:
Ok, I think we're ready to rock. I've done as much testing as possible, but there's still a high probability that I've missed something. Are you guys keen to pull down the 0.1.3 branch that I have pending?

https://github.com/davidmckenzie/pagermon/tree/0.1.3

I've put up some upgrade instructions here: https://github.com/davidmckenzie/pagermon/wiki/0.1.3-Upgrade-Instructions

marshyonline, 2017-06-29:
Ill pull it down tonight and give it a crack :)

jklnz, 2017-06-29:
That was nice and easy then (lucky!).
I'm about to break it out and test it right now

jklnz, 2017-06-29:
Just out of curiousity how are you running the sqlite commands? Is running through through sqlitebrowser ok?

davidmckenzie, 2017-06-29:
I tested in DB Browser for SQLite on OSX, and in the linux sqlite3 command. :)

Just noticed some code from previous commits to fix the websockets issue is missing... not sure how that happened, will pop it back in now though.

jklnz, 2017-06-29:
All good, I started reading your wiki and found the answer

Bugger, just downloaded it, haha, let me know you've done it and i'll re-download

davidmckenzie, 2017-06-29:
Go for it, just pushed the fix. :)

jklnz, 2017-06-29:
Sweet, downloading now.

FYI just run all the DB commands and no errors so looking good

jklnz, 2017-06-29:
Looks like it worked, I must've done something wrong as it reset my API keys to the defaults. But otherwise looking good

jklnz, 2017-06-29:
Feels alot faster, nice work

jklnz, 2017-06-29:
And I see you fixed the glitch where it did not pre-load the capcode when clicking to add aliases, that's great

jklnz, 2017-06-29:
I know what I did, i removed the old config folder instead of keeping it

[12 more replies]

commit 5063fbcd (2017-06-29): Quick fix #43
closed 2017-07-10 by marshyonline