pagermon/pagermon #44: Suggestion: Auto Close "Alias Saved" green box after you've saved an alias

jklnz, 2017-06-23:
If that is possible?

davidmckenzie, 2017-06-24:
Have pushed a change that does this now - in my testing it was a little unpredictable, probably due to the dodgy way I implemented the message displaying. Hopefully works alright though :)

jklnz, 2017-06-30:
Hey @davidmckenzie - Just letting you know I dont think this worked, I still have the green box stay up after editing / adding Aliases

davidmckenzie, 2017-06-30:
Huh weird... it's working on dev but not on prod for me.

davidmckenzie, 2017-06-30:
Ok I see the issue - it fades once, but if you perform another action that generates a message without changing routes (i.e. click save, then click refresh aliases), then it doesn't fade.

jklnz, 2017-06-30:
I was just going to write that I could get it to disappear sometimes, but not others - guess that explains it

davidmckenzie, 2017-07-23:
Moved the timeout function out to the things that are actually triggering the alerts, that should do it. Only downside is the alerts don't fade away all pretty-like, but a worthy trade-off ;)

The code controlling those alerts is reaaaally messy. Need to pop it out to a function sometime soon.

commit 3e69d853 (2017-06-24): Resolves #44
commit cfa0bbb0 (2017-07-23): Fixed alert close #44
commit cfaed022 (2017-07-23): Merge pull request #59 from davidmckenzie/hide-alerts

Fixed alert close #44
closed 2017-07-23 by davidmckenzie