pagermon/pagermon #45: Ignoring POCSAG2400 ?

booxdk, 2017-06-23:
Hi. I have everything set up and running.
Launching testMessage.sh is succesfull, but I'm not getting any messages from the POCSAG2400 that I'm receiving in my area ( my receiving setup works piping to a txt-file)
If I edit the testMessage.sh to test with "pocsag2400" like this, i don't receive the test message.
`
#!/bin/bash

cd .. && echo "POCSAG2400: Address: 1000000  Function: 0  Alpha:   Test message, disregard<EOT>" | node ./reader.js
`

Duplicate filtering is disabled, and sending the POCSAG512 test message multiple times works every time.

BooX

booxdk, 2017-06-23:
Just looked into the code of reader.js and realized that it only supports POCSAG512 and FLEX at the moment.
I've done a quick edit changing the POCSAG512 part to POCSAG2400 without thinking deeper into the rest, and the POCSAG2400 test message now shows up.

I'm testing with actual messages at the moment. There isn't much traffic here ( only fire department ) so I will have to wait  for messages, and will post the result as soon as I get som messages in to test with.

BooX

davidmckenzie, 2017-06-23:
Thanks man, have pushed a quick change to use regex instead of indexof, should work on the other types of POCSAG :)

Haven't tested thoroughly, but should work... in theory

booxdk, 2017-06-24:
I'm the one saying thanks a lot :-) Love your great work !
I've just updated the files to #46 and was lucky that the daily test message was sent 5 minutes later.
I can confirm that POCSAG2400 works with this version - great !

My frontpage of Pagermon still says PagerMon 0.1.0-alpha in the lover left corner, did you update the version ?

BooX

davidmckenzie, 2017-06-26:
Cos I'm super smart, I put the version number in the config file... which doesn't get updated when git updates.

Fix will be in the next version :D

booxdk, 2017-06-26:
I can live with the incorrect version display as long as I know that I didn't fail to update the files correctly :o)))
Btw... I have received a lot of POCSAG2400 since last post, and it works great.

Thank you for your great work :o)

closed 2017-06-26 by booxdk