pagermon/pagermon #55: Message failed to deliver. StatusCodeError: 401 ...Authentication Failed  ???

chudgoo, 2017-07-06:
Is some auth required for the reader.js to autheticate to the Sqlite3 server? 
I can't seem to find any place in the script that mentions auth.

davidmckenzie, 2017-07-06:
Yup - once the server is started, you need to create an API key in the admin panel under 'settings'.

That said, I'm getting errors on my install at the moment trying to navigate there - it seems the 'colorpicker' include is broken at the moment:

https://unpkg.com/angularjs-color-picker@3.3.0/dist/angularjs-color-picker.min.js

I'll switch to the non-minified include of that until unpkg fixes it :)

davidmckenzie, 2017-07-06:
And fixed :) Might wanna pull down the latest version if you're seeing nothing in the settings page.

So what you'll wanna do, is point to the server in your browser and login with the default credentials (admin / changeme), change the password through the 'reset password' page, then go to the settings page and right down the bottom is the API Key settings. Hit the plus sign to create a new key, then click 'generate' and it'll give you a nice long key that you can paste into the reader.

Lemme know if you have any trouble :)

chudgoo, 2017-07-06:
AWESOME!  That was the trick.  Thanks so much for this project. I was trying to do something very low level in this vein for a while but never got passed just piping everything out to /var/www/html/index.html.... so bad.  just one huge file that had a header that auto reloaded the file every 5 minutes.... pagermon (now that I've got it running) is like a dream come true....  ;)

davidmckenzie, 2017-07-06:
Good stuff 😄 hopefully all goes smoothly from here!

chudgoo, 2017-07-07:
One more quick question... What is the prefered method (in Raspbian) to lauch reader.sh to start at boot?
Adding the script to /etc/init.d/ doesn't seem to work.  It starts 'rtl_fm | multimon-ng' but not the 'node' part...  Any thoughts?

davidmckenzie, 2017-07-08:
Hmm not too sure - I haven't done any testing on my Raspi with the reader just yet. Perhaps run it with nohup or screen? Screen might be best, since you can jump into the screen session if you want to view the pretty coloured output.

chudgoo, 2017-07-08:
Ha! I ended up doing tmux (same idea as screen basically) and closed the ssh session. 
It should just run and run now.  I will continue looking into ways to get the ENTIRE string of commands to execute on launch.  Currently it's still just rtf_fm and multimon-ng that fire off. 'node reader.js' never seems to execute. I even made sure the path to the node was in my bashrc... weird.

davidmckenzie, 2017-07-23:
Just a thought - have you tried using nodemon? I dunno if it accepts piped inputs, if so might be a little more stable on boot :)

kiwiscout, 2017-12-27:
Probably a bit late to the game to answer the 2nd portion of your query now.., 

But after just shifting from PDW to Pagermon myself and spending way more time than I probably should (in excess of probably 6hrs) getting it working flawlessly I too struck this and couldn't work out how to pipe the rtl_fm/multimon output to a background active process in a single step until I found pm2 supports .sh like bash. So I've got both auto-loading via pm2 and then saved to startup per the server install instructions. sudo pm2 start reader.sh (from client folder) and from pagermon root the normal sudo pm2 start server\process.json

closed 2018-10-11 by marshyonline