pagermon/pagermon #56: Find and replace text?

chudgoo, 2017-07-07:
Is there currently a method to find text in a page like "<NUL>" and replace it with blank space/no character?

chudgoo, 2017-07-07:
weird. the comment was altered...  < N U L > was what was supposed to be between those quotes.

davidmckenzie, 2017-07-08:
Only by modifying the reader.js - take a look at line 67 on there:

```
trimMessage = message.replace(/<EOT>/g,'');
```

We could replace that with:

```
trimMessage = message.replace(/<EOT>/g,'').replace(/<NUL>/g,'');
```

(That's as long as it's POCSAG - for flex, you'd do something similar at line 76.)

Perhaps it would be smarter for us to expand the regex so it matches anything with 3 chars between two pointy brackets... e.g.

```
trimMessage = message.replace(/<[A-Za-z]{3}>/g,'');
```

Not sure what potential codewords would be in there though... may have to do some digging.

chudgoo, 2017-07-08:
that " < N U L > " thing is the only bit of garbage text I see with every POCSAG message, and after swapping out line 67, everything has cleaned up nicely.  Thanks for the prompt attention to my minor little issue.  ;)

commit ff52a21e (2017-07-23): Resolves #56
closed 2017-07-23 by davidmckenzie