pagermon/pagermon #60: Bulk add hundreds of aliases?

chudgoo, 2017-08-04:
If someone had a CSV file of all the capcodes and associated users, it would be nice if there a way to bulk add them to the PagerMon database.  If this is already possible, I apologize for making a new issue of it.

davidmckenzie, 2017-08-04:
That's kind of a thing in the import.js in the client dir - it just parses a filters.ini file from PDW. It is in CSV format, so you could either format your CSV to match, or modify the import script to select different columns :)

https://github.com/davidmckenzie/pagermon/blob/master/client/samples/filters.ini

https://github.com/davidmckenzie/pagermon/blob/master/client/import.js

A lot of the import.js script is based on my personal use of the filters.ini, so generally does need a little modification to suit local envs. E.g. I assume that the alias will be in the format of "Agency - Brigade/Unit"

davidmckenzie, 2019-03-04:
Reopening this as I'd like to create a generic csv importer

davidmckenzie, 2019-04-26:
Closing - in 0.3.0 release

PR #260 (2019-04-26): Import csv
# Description

Updates import.js to accept generic CSV files

Fixes #60

## Type of change

Please delete options that are not relevant.

- [x] New feature (non-breaking change which adds functionality)

# How Has This Been Tested?

I imported a buttload of aliases. It went ok.

It posts very aggressively, so I had a few failures on import. Need to slow it down and make it retry, but "good enough" for now.

# Checklist:

- [x] My code follows the style guidelines of this project
- [x] I have performed a self-review of my own code
- [lol] I have commented my code, particularly in har [... cut]
closed 2019-04-26 by davidmckenzie