pagermon/pagermon #63: Feature requests

oh2ftu, 2017-09-14:
Here are some features that might become useful:
- Integrate rtl_fm control to the client via xmlrpc
- Add possibility to change timestamp format (from HH:mm to HH:mm:ss)
- Add script to run per alias, or perhaps GET or POST -request? might also require a hysteresis here
- Message parser, ie. if an address is given, it becomes a link to google maps or equivalent.
- Documentation to getting started might be a bit improved. I got there eventually :)

davidmckenzie, 2017-09-14:
These are awesome suggestions, thanks! I sadly haven't had much time for this project lately, work commitments getting in the way. More than happy to review any PRs if anyone else gets to these in the meantime. :)

davidmckenzie, 2018-11-19:
FYI your 3rd suggestion is partially being solved in #130 :)

eopo, 2019-09-30:
To your issue 4 - I have a custom written plugin that can do the parsing (And theoretically do the google maps link as well, if I'd implement it). I could share the code with you so you can customize it for your needs.