pagermon/pagermon #64: Numeric Pages

marshyonline, 2017-09-19:
Is the reader ignoring Numeric pages?

[image]

Is is a numeric only page, but as you can see it has only passed a junk Alpha message to the server.

marshyonline, 2017-09-19:
Another example. somthing stange going on with it, I cant work out why im getting all these bad numeric and skyper messages with every single message what comes thu 

[image]

davidmckenzie, 2017-09-19:
Yeah I filtered out numeric originally as it seemed to be largely junk data, can definitely see it being of use though. Would just need to duplicate line 65:68 in reader.js, should be easy :)

What command are you running with on the multimon side? Tweaking the '-b' flag can reduce a lot of false positives

davidmckenzie, 2017-09-19:
Just pushed a commit to allow the numeric pages. I left the filter regex in there (the one that kills off things like <DEL>), not sure if it's required for numeric though. Let me know how that goes, I don't have much numeric to test with around here, only a daily test page :)

marshyonline, 2017-09-19:
Stange
The messages that appear to be blank outputted by the reader are actually not, just black text 

Received: POCSAG1200: Address:  603410  Function: 0  Alpha:   08:40 P90390..HUTMSG.MON.F1<NUL>
2017-09-19 22:40:57: 603410: 08:40 P90390..HUTMSG.MON.F1
Received: POCSAG1200: Address:  603410  Function: 0  Skyper:  /793/<US>O8/28/--GTSLRF-LNM-E0�
2017-09-19 22:40:57: POCSAG1200: Address:  603410  Function: 0  Skyper:  /793/<US>O8/28/--GTSLRF-LNM-E0�
Received: POCSAG1200: Address:  525886  Function: 2  Numeric: 03 9]86030-4U81505 071391 4600
2017-09-19 22:40:58: POCSAG1200: Address:  525886  Function: 2  Numeric: 03 9]86030-4U81505 071391 4600
Received: POCSAG1200: Address:  525886  Function: 2  Alpha:   08:40 SEQ 181202<NUL>
2017-09-19 22:40:58: 525886: 08:40 SEQ 181202
Received: POCSAG1200: Address:  525886  Function: 2  Skyper:  /793/<US>RDP<US>0701/1�
2017-09-19 22:40:58: POCSAG1200: Address:  525886  Function: 2  Skyper:  /793/<US>RDP<US>0701/1�
Received: POCSAG1200: Address: 1601152  Function: 2  Numeric: 03 9]86130145653.5 2663714380582-00
2017-09-19 22:41:33: POCSAG1200: Address: 1601152  Function: 2  Numeric: 03 9]86130145653.5 2663714380582-00
Received: POCSAG1200: Address:  603410  Function: 0  Numeric: 03 9]8613014370U95]06]2712U.4-6]4[8].6]3-9]236 000
2017-09-19 22:41:57: POCSAG1200: Address:  603410  Function: 0  Numeric: 03 9]8613014370U95]06]2712U.4-6]4[8].6]3-9]236 000
Received: POCSAG1200: Address:  603410  Function: 0  Alpha:   08:41 P90390..HUTMSG.MON.F1<NUL>
2017-09-19 22:41:57: 603410: 08:41 P90390..HUTMSG.MON.F1
Received: POCSAG1200: Address:  603410  Function: 0  Skyper:  /7930<US>O8/28/--GTSLRF-LNM-E0�
2017-09-19 22:41:57: POCSAG1200: Address:  603410  Function: 0  Skyper:  /7930<US>O8/28/--GTSLRF-LNM-E0�
Received: POCSAG1200: Address:  525886  Function: 2  Numeric: 03 9]86130-4U81505 071391 6600
2017-09-19 22:41:58: POCSAG1200: Address:  525886  Function: 2  Numeric: 03 9]86130-4U81505 071391 6600
Received: POCSAG1200: Address:  525886  Function: 2  Alpha:   08:41 SEQ 181203<NUL>
2017-09-19 22:41:58: 525886: 08:41 SEQ 181203
Received: POCSAG1200: Address:  525886  Function: 2  Skyper:  /7930<US>RDP<US>0701/2�
2017-09-19 22:41:58: POCSAG1200: Address:  525886  Function: 2  Skyper:  /7930<US>RDP<US>0701/2�

marshyonline, 2017-09-19:
Or look very closely at this image 
https://imgur.com/a/TBQly

davidmckenzie, 2017-09-19:
Oh right! That colour is the ANSI "grey" colour, which depending on shell settings may appear quite difficult to see.

```
  // filter out most false hits
  // if too much junk data, make sure '-p' option isn't enabled in multimon
  if (address.length > 4 && message) {
    var padAddress = padDigits(address,7);
  	console.log(colors.red(time+': ')+colors.yellow(address+': ')+colors.success(trimMessage));
...
  } else {
  	console.log(colors.red(time+': ')+colors.grey(line));
  }
```

So anything where the address is less than 4 chars long, or where the script doesn't set the message (things that don't match the 'Alpha:' regex or 'Numeric:' as well now) will output with the line with the muted text as an indicator that the pager received something but it was ignored.

marshyonline, 2017-09-19:
Ahh ok makes sense now! 

I have tryed changing the -b to B1 B2 B3 and b0 but no anvil, all garbage numeric and Skyper message with every legit alpha and the other way around for numeric messages 
Strange indeed

davidmckenzie, 2017-09-19:
'-u' might help too... or could be something simple like your sdr has started drifting a bit more than usual :)

marshyonline, 2017-09-19:
I have tryed -u but it just misses way too many messages. 
The SDR hasn't drifted so I think it may just be a reception issue. Am under 10kms from the transmitter so it could be overloaded 

Another suggestion is to not allow any alpha or numeric message under 3 chars long

marshyonline, 2017-09-19:
I need to sort out this garbage issue before I can test, otherwise it's going to flood the system with crap haha

marshyonline, 2017-09-20:
I have come up with an idea to solve my specific issue 
Filter numeric messages out that are larger then 5 characters
And like wise for alpha messages shorter then 5 characters
This has to happen after messages are trimmed also 
Any ideas on how to implement ?

davidmckenzie, 2017-10-11:
Sorry dude, I completely forgot about you again :(

I can't test, but if you wanted to do something like this on line 63:74 of reader.js it might work:

```
  if (/^POCSAG(\d+): Address: /.test(line) ) {
  	address = line.match(/POCSAG(\d+): Address:(.*?)Function/)[2].trim();
    if (line.indexOf('Alpha:') > -1) {
    	message = line.match(/Alpha:(.*?)$/)[1].trim();
    	trimMessage = message.replace(/<[A-Za-z]{3}>/g,'');
    	if (trimMessage.length < 5) {
    	    	// filter out junk
    	    	message = false;
    	    	trimMessage = '';
    	}
    } else if (line.indexOf('Numeric:') > -1) {
        message = line.match(/Numeric:(.*?)$/)[1].trim();
        trimMessage = message.replace(/<[A-Za-z]{3}>/g,'');
    	if (trimMessage.length > 5) {
    	    	// filter out junk
    	    	message = false;
    	    	trimMessage = '';
    	}
    } else {
    	message = false;
    	trimMessage = '';
    }
```

If that's all good I can pop a commit in :)

marshyonline, 2017-10-12:
Allgood i got stuck into it and wrote basically  the same thing, i need to
make a small  chance and ill post tonight

marshyonline, 2017-10-22:
I went down this path.
https://gist.github.com/marshyonline/c711ce9c906d8fa23c6023637a18f9a6

Ive tweeked and tweaked the numbers time and time again, and this seems to be the best out of the lot. No missing messages so far with no false messages being sent to the server

closed 2018-07-07 by marshyonline