pagermon/pagermon #66: Pushover Priority Per Alias

thewillhargan, 2017-10-04:
Got the program up and running and it's great, but I'd really like to be able to push different aliases at different priorities.

For Example. I'd like to be able to push any messages that would usually go to my pager at Emergency Priority. Whereas other pages can be sent at normal priority

davidmckenzie, 2017-10-11:
Would need DB structure change unfortunately - will have to keep this one on the backlog for when I eventually get around to redoing the DB side of things. Will be looking at backing it on MongoDB or DynamoDB :)

marshyonline, 2017-10-22:
This is something i never thought of when i integrated pushover initially, but i really do like this idea. 
But David is right, its not something i would want to push out as it will require another database change and that's something I want to avoid doing again.
+1 on this idea, keen to see it in the newer versions

marshyonline, 2018-04-23:
I have made this change in #74 :)

closed 2018-04-23 by davidmckenzie