pagermon/pagermon #70: [] in message used as capcode when present

kiwiscout, 2018-01-08:
Some messages received here contain information status where the sender reports in []s. For example middle of message [Situation: ... ] and is mis-interpreted as a capcode and most of the message until next break char is put into the capcode column

Checking, I can likely fix this by ammeding the [] from address matching but remain in message?

 else if (line.indexOf('FLEX: ') > -1) {
    address = line.match(/FLEX:.*\[(.*?)\] /)[1].trim();
    if (line.match( /( ALN | GPN | NUM)/ )) {
      message = line.match(/FLEX:.*\[.*\] ... (.*?)$/)[1].trim();
      trimMessage = message;

davidf2002, 2018-02-24:
I've noticed the same problem. I think it's the regexp in reader.js to determine the address.

eg. for a pager message that looks like:
FLEX: 2018-02-24 03:15:28 1600/2/K 03.111 [001900157] ALN FRAN (897,0) Antenna Power Low [AntL] SN:693669 ST:AL LC:1 02/24/18 12:40:14

reader.js (around line 76, I've added some debug code so my line numbers might be out) determines the address to be  "AntL" instead of "001900157".

I think the regexp:    address = line.match(/FLEX:.*\[(.*?)\] /)[1].trim();
just needs to be changed to match the first occurance of [] not the last.

I tried changing it to:   FLEX:(.*?)\[(.*?)\] 
but it didn't work.

kiwiscout, 2018-02-24:
Indeed, I had numerous edits of the code as examples also that the editor didn't like. I too tried the same change and it wasn't happy at all. 

Not familiar enough with jscript to analyze and adjust the line.match code to make it check only the first chr(9) or similar myself.

kiwiscout, 2018-02-24:
Possibly have a solution using the numerical only selector within the []s. Replacing .*? with .*\d

On my input test at regex101.com it only matched the capcode and not subsequent. I need to get a message to test if it crashes now

.*\[(.*\d)\]

davidmckenzie, 2018-02-25:
\d or [0-9]* could work, but would have the same issue if the text further on contained only numbers.

Will have a quick play with the regex now :)

davidmckenzie, 2018-02-25:
Ohhh, @davidf2002 you were super close - making it non-greedy was the right path, but because you encased it in brackets, that means that the first bit would've been the first item in the array.

/FLEX:.*?\[(.*?)\] / should work, and we can use \d as well for extra safety. Will update now!

davidmckenzie, 2018-02-25:
Let me know how that goes, hopefully all good now :)

davidf2002, 2018-02-25:
thanks guys, that looks like it should fix it, I've been waiting over 24hrs for another page in this format to test it, still waiting....

kiwiscout, 2018-02-25:
Silly github.. it truncated what I adjsted to (which I think is the same change DM has now made with brackets to target the first instance only like it was. But I only added to the address section, not message

`address = line.match(/FLEX:.*\[(.*\d)\] /)[1].trim();
    if (line.match( /( ALN | GPN | NUM)/ )) {
      message = line.match(/FLEX:.*\[.*\] ... (.*?)$/)[1].trim();`

I had one today with additional [] message that didn't throw part of it in the capcode field... so looks like it'll work targeting first in array only.

davidf2002, 2018-02-25:
got another one of those pages, looks like it's working to me, cheers!

closed 2018-10-06 by davidmckenzie