pagermon/pagermon #72: use timestamp in pager message instead of current time

davidf2002, 2018-02-24:
reader.js grabs the current time but FLEX pages at least have the time in the pager message and it would be great to use that instead.  I have multimon-ng pipe into some other programs for process, upload to AWS etc and the time is often out by the time it gets to pagermon reader.js.  

FLEX pages looks like:
FLEX: 2018-02-24 05:07:55 1600/2/K 01.125 [001900253] ALN THIS IS A TEST PERIODIC PAGE SEQUENTIAL NUMBER  9556

(note that time is UTC, so do a localtime() on it)

davidmckenzie, 2018-07-06:
Little late but just thinking about this one, since not all pager formats will show a time (and the time may not always be valid if decoding fails for some reason), I wonder if a better solution for your specific problem might be FIFO pipes in the bash script itself. Something like the second answer here: https://stackoverflow.com/questions/60942/how-can-i-send-the-stdout-of-one-process-to-multiple-processes-using-preferably

So you'd create a FIFO pipe for the AWS upload and for the reader.js, then start those processes separately reading from the pipes.

davidmckenzie, 2018-11-05:
Closing, will partially solve this in #57

davidmckenzie, 2019-03-04:
Reopening this as I'm looking at doing something similar - my timestamps are in format `05 March 2019 07:55:52`. I'd like to strip it from the message and use it as the metadata, to reduce dupe messages.

I'll see if there's some more generic way of detecting many timestamp formats, if not I'll make it work for your timestamp format and mine.

Will be an optional flag or config option on the reader.js script to strip/use these.

davidmckenzie, 2019-03-17:
Looking at this now, got the following regex, I don't really like it but I think it should cover the known cases.

@pagermon/collaborators have you guys seen any other datetime formats floating around?

[image]

davidmckenzie, 2019-03-18:
Moment doesn't like my timestamps, so I'll have to split these out individually anyways :(

PR #235 (2019-03-18): Timestamps
# Description

Parses known timestamp formats and strips them from the message, using them instead to form the timestamp metadata. This reduces duplicates in poorly configured pager environments (like NSW).

The regex might be a bit greedy, will have to see how it plays live

Fixes #72 and #57 (probably)

## Type of change

Please delete options that are not relevant.

- [x] New feature (non-breaking change which adds functionality)
- [x] This change requires a documentation update

# How Has This Been Tested?

Tested in sample test script with FLEX and POCSAG messages with and with [... cut]
closed 2019-03-18 by davidmckenzie