pagermon/pagermon #75: Enhancement - Change Alias settings for all Aliases assigned to Agency

Lowestkillcount, 2018-06-20:
Hi David,

Great project you have here!

I''m thinking it would be good to be able to update all aliases assigned to an agency at once - possibly have them as templates so you don't need to set the settings on every alias?

I'd assist and do it myself, but unfortunately i'm a powershell man and this is all gobbledygook to me :)

Cheers
Dan

therokh, 2019-02-06:
I think creating a new object for Agencies and linking aliases to an Agency is a good method. 
Each Agency can be configured with a colour and icon that is inherited by all assigned Aliases. Aliases can override the inherited colour and icon but still be attached to an Agency.

Something like this:
`Agency - MFS (colour=red, icon=fire-extinguisher)`
` | - Alias - MFS Station 1 (inherit above)`
` | - Alias - MFS Truck 34 (inherit colour, override icon=truck)`

DanrwAU, 2019-02-06:
Yep, that was my thoughts too, i've just been too lazy to implement 💥 

Maybe something ti look at after 0.3.0