pagermon/pagermon #76: Queuing pager messages on client machine when server is not contactable.

s3m1s0n1c, 2018-06-20:
Thank you for this awesome program,

Its a shame i've only just discovered it as i've been using another program but like this one as u can do filtering and it deals with duplicates more cleanly.

Does Pagermon have some sort of buffer/queuing system if the server is not able to be contacted?

Eg If Internet drops from client and it receives a pager message then it drops the pager messages cause it cant contact the server.

Solution is to have some sort of file it send messages too and reader.sh reads off it and deletes the row when finished.

Moving to SQL DB would solve this also as u can sync DB's which is what I do atm.

Food for thought.. Maybe this can be a feature one day.

davidmckenzie, 2018-06-21:
Could add some looping in the catch on line 114, if it fails then send it off to a setTimeout driven function that retries every x seconds. Might lead to memory leakage if the server is down for a while though.

davidmckenzie, 2019-01-13:
Closing as solved in #173, I'd like to revisit this at some point though

closed 2019-01-13 by davidmckenzie