pagermon/pagermon #77: API Security

s3m1s0n1c, 2018-06-21:
API calls from /api and /post need to be secured a bit more.

Anyone can pull all the info from the DB eg. Capcodes, Messages etc using http://ip:3000/api/messages or /api/capcodes

I think its important to keep data secure.

Thanks
Sonic

s3m1s0n1c, 2018-06-21:
**### Temp Fix**

A temp fix is to edit the following files

**server/app.js**

change line 144 from
app.use('/post', api);
to
app.use('/X', api);

Change X to anything you want.. it doesn't matter as long as its not post. I don't think this function get used in app.js and is only for API Calls from external programs.

change line 145 from
app.use('/api', api);
to
app.use('/Y, api);

Change Y to anything you want.. eg api8375 or 01api92

save file

**client/reader.js**

change line 33 from
var uri = hostname+"/api/messages";
to
var uri = hostname+"/Y/messages";

save file

**public/javascripts/admin/admin.main.js**

change line 6 from
Aliases: $resource('/api/capcodes/', null, {
to
Aliases: $resource('/Y/capcodes/', null, {

change line 9 from
AliasDetail: $resource('/api/capcodes/:id', {id: '@id'}, {
to
AliasDetail: $resource('/Y/capcodes/:id', {id: '@id'}, {

change line 18 from
AliasDupeCheck: $resource('/api/capcodeCheck/:id', {id: '@id'}, {
to
AliasDupeCheck: $resource('/Y/capcodeCheck/:id', {id: '@id'}, {

change line 21 from
AliasRefresh: $resource('/api/capcodeRefresh', null, {
to
AliasRefresh: $resource('/Y/capcodeRefresh', null, {

save file

**server/views/index.ejs**

change line 159 from
Messages: $resource('/api/messages/'),
to
Messages: $resource('/Y/messages/'),

change line 160 from
NewMessage: $resource('/api/messages/:id', {id: '@id'}),
to
NewMessage: $resource('/Y/messages/:id', {id: '@id'}),

change line 161 from
MessageSearch: $resource('/api/messageSearch/')
to
MessageSearch: $resource('/Y/messageSearch/')

**You will have to restart your server using pm2 restart pagermon**

davidmckenzie, 2018-06-21:
Yeah that change won't really help much - the routes are exposed in the web app. The messages routes *have* to be exposed for the app to work, all you're doing there is making it so that whoever wants to use them has to spend 10 seconds in Chrome dev tools looking at what requests are made.

You could probably add some security to the capcode routes if you're concerned about people stealing your capcode lists.

Personally I don't support locking that down as it goes against my philosophy of open information sharing, but if you wanted to put in a PR for that then I'd be happy to take a look.

Alternatively, if you're concerned about people accessing *anything* then I'd suggest chucking some basic auth on nginx or whatever you're using for proxying.

s3m1s0n1c, 2018-06-23:
Oh okays damm.. Yeah, I'm no expert in Node Js but yeah I can seem them in chrome dev tools :(

How would I add security to capcodes route?
How do i put in a PR?

Its just taking me a long time to get them and there not public knowledge otherwise everyone would decode them.

Thanks
Sonic

s3m1s0n1c, 2018-06-26:
Will this work?

Added the following to line 50 of routes/api.js

router.get('*',
  passport.authenticate('localapikey', { session: false, failWithError: true }),//
    function(req, res, next) {
      next();
  },
  function(err, req, res, next) {
      console.info(err);
      isLoggedIn(req, res, next);
  });

But not sure how to call the API key in views/index.ejs as messages don't display cause obviously there is no key being sent.

Will the API key be publicly shown anywhere if I do it this way?

Thanks for your help

DanrwAU, 2018-09-15:
This pull request should give you what you wanted if still required :wink:

https://github.com/davidmckenzie/pagermon/pull/92

closed 2018-09-17 by davidmckenzie

s3m1s0n1c, 2018-09-18:
Thanks Guys for all your work..

Works great :)

Thanks Sonic