pagermon/pagermon #84: Searching times out

mysteriousbrad, 2018-07-30:
Hi
Have been running pagermon for a month now on a raspberry pi 2 (dietpi OS). 
Now that my database has become quite sizable (20M) I can no longer search for anything, as doing so causes loading ring to come up for a minute or so, then goes back to the search screen with no results, when I know there would be results. Clicking on a capcode works fine, and brings up listings straight away.
Wondering whether theres just a timeout that needs to be extended, database need to be culled, or am I asking too much of the raspberry pi 2?

Kind Regards

Brad

davidmckenzie, 2018-07-30:
Probably the last two :) Anything in the logs (server side and on the client side web console)?

DanrwAU, 2018-07-30:
For what it's worth, i tried running it locally on my Pi 3b (also dietpi) and had issues with speed with my DB Size (36k messages, 1200 Aliases) - would max the CPU out pretty easily with Rtl_FM/Multimon etc all running. 
I threw up a VPS server in Azure and pipe everything to that, works perfectly.

mysteriousbrad, 2018-07-30:
When I do search, the following goes into stdout logs:

2018-07-30 14:08 +10:00: ::ffff:192.168.188.31 - - [30/Jul/2018:04:08:48 +0000] "GET /api/messages?limit=&page=1 HTTP/1.1" 200 - "http://192.168.188.41:3000/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/64.0.3282.140 Safari/537.36 Edge/17.17134"
2018-07-30 14:08 +10:00: ::ffff:192.168.188.31 - - [30/Jul/2018:04:08:55 +0000] "GET /?q=ROCK HTTP/1.1" 200 - "http://192.168.188.41:3000/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/64.0.3282.140 Safari/537.36 Edge/17.17134"
2018-07-30 14:08 +10:00: ::ffff:192.168.188.31 - - [30/Jul/2018:04:08:55 +0000] "GET /stylesheets/style.css HTTP/1.1" 304 - "http://192.168.188.41:3000/?q=ROCK" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/64.0.3282.140 Safari/537.36 Edge/17.17134"
2018-07-30 14:08 +10:00: init: 0.498ms
2018-07-30 14:09 +10:00: Listening on port 3000
2018-07-30 14:09 +10:00: ::ffff:127.0.0.1 - - [30/Jul/2018:04:09:15 +0000] "POST /api/messages HTTP/1.1" 200 9 "-" "-"
2018-07-30 14:09 +10:00: sql: 23264.098ms
2018-07-30 14:09 +10:00: ::ffff:127.0.0.1 - - [30/Jul/2018:04:09:19 +0000] "POST /api/messages HTTP/1.1" 200 9 "-" "-"
2018-07-30 14:09 +10:00: init: 1.919ms
2018-07-30 14:09 +10:00: sql: 23194.396ms

Don't get any error messages on the website just the search page with the term in the search box and nothing underneath as if nothing matched

mysteriousbrad, 2018-07-30:
Thanks @DanrwAU, def worth thinking about!

DanrwAU, 2018-07-30:
Just ran the same search on my DB and it takes 5 seconds to complete even on an 2CPU Azure VM.

I too have plenty of results :)

2018-07-30 04:33 +00:00:  - - [30/Jul/2018:04:33:36 +0000] "GET /?q=ROCK HTTP/1.0" 200 - "https:///" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/68.0.3440.75 Safari/537.36"
2018-07-30 04:33 +00:00:  - - [30/Jul/2018:04:33:36 +0000] "GET /stylesheets/style.css HTTP/1.0" 304 - "https:///?q=ROCK" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/68.0.3440.75 Safari/537.36"
2018-07-30 04:33 +00:00: init: 0.041ms
2018-07-30 04:33 +00:00: sql: 679.234ms
2018-07-30 04:33 +00:00: search: 0.874ms
2018-07-30 04:33 +00:00: searchFullText: 5319.245ms
2018-07-30 04:33 +00:00: sort: 0.365ms

davidmckenzie, 2018-07-30:
That 23 seconds in sql is probably it:

2018-07-30 14:09 +10:00: sql: 23264.098ms

My money would be on memory contention or swapping. The SQL part of the query can be heavy, but not usually *that* heavy :)

You could try looking at some [memory reduction things for sqlite](https://stackoverflow.com/questions/15255409/how-to-reduce-sqlite-memory-consumption), but might just be delaying the inevitable. Having separate boxes for the reader and the server parts is usually best.

mysteriousbrad, 2018-07-30:
Thanks @davidmckenzie - I have access to a VPS I can put nodejs on, so will give that a try!
Also thanks for the heads up on SDRTrunk, will check that out!

mysteriousbrad, 2018-07-30:
Working much better with the server instance on vps! Thanks guys.

closed 2018-07-30 by mysteriousbrad