pagermon/pagermon #89: Seperate Repos for components

DanrwAU, 2018-08-17:
Hey Guys, 

Would it make sense to split out the components to individual repositories?

This would allow running of the server app on platforms like Heroku much simpler and prevent having to clone the entire server to platforms that will only be run as a client (and vice versa). 

Possibly something like pagermon-server, pagermon-client, pagermon-docker? 

Just a random middle of the night thought :)

davidmckenzie, 2018-08-19:
Yknow it probably wouldn't be a bad idea. Separate versioning kinda makes sense with the client side too since it's a lot more flexible - it can afford to be a bit messier with people adding different client side scripts, lot looser merge rules.

Lemme have a think about how best to implement that. I'd really like to get back into some of the big ticket items in the backlog when time permits, can start looking at it with that :)

damiankw, 2018-10-09:
So you don't have to lose what you have here, and you can just build on it, you could potentially just fork the repository into the different names, delete what isn't required and save back. It means history will be there for everything, even if it's not valid, but from this point on everything is individual to the actual element.

From this primary repository, you could add all of the others in simply by doing a [submodule](https://git-scm.com/book/en/v2/Git-Tools-Submodules), that way if people want to download the entire package they can clone this repository, then pull each of the modules individually.

DanrwAU, 2019-05-10:
@davidmckenzie had a play with submodules. How do you feel about this? 
https://github.com/DanrwAU/pagermon/tree/Cleanup