pagermon/pagermon #90: Help Needed

berto547, 2018-08-18:
Hi, Not an issue with the program just an issue with me. I am having trouble setting it up. If anyone could help that would be appreciated. email fireext@gmail.com Thanks

DanrwAU, 2018-08-18:
What OS are you trying to deploy on?
Which step are you having trouble with?

berto547, 2018-08-18:
I have Windows. I have setup a server, I think but can't see anything when I open browser. Not sure what to put into process.json using the git terminal it says the status is online.  I am also using power shell to get the data from PDW. PDW I have setup no problems. Thanks for the reply.

berto547, 2018-08-18:
I have Windows. I have setup a server, I think but can't see anything when
I open browser. Not sure what to put into process.json using the git
terminal it says the status is online. I am also using power shell to get
data from PDW. PDW I have setup no problems. Thanks for the reply.

DanrwAU, 2018-08-18:
So your deploying the server component on windows?

Where exactly are you getting stuck. Unfortunately we really do need allot more information to be able to assist

berto547, 2018-08-18:
I have the server up and running now. Have logged in to admin.
How do I generate an API
Can't seem to see any messages.
Have saved psclient.ps1 to c drive.
Haven't put in an api key as not sure where to generate it.
App url /api/messages I can't even find that location.
Think I am close just need to config a few things

DanrwAU, 2018-08-18:
OK then. 

Generating an API is done on the settings page, simply delete the default ones and generate a new one. 
https://i.imgur.com/IarteqL.png

The powershell client requires you to update the following lines
$sourcename 	= "***Source name***" This can be anything you like - i use the hostname of the device that posted the message

$appurl 	= "http://***App URL***/api/messages" This is the address you are using in your web browser 
if you are using http://192.168.0.10:3000 to access, this becomes http://192.168.0.10:3000/api/messages

$apikey 	= "***API Key***" This is the API Key you generated at the start. 

You then setup PDW to run a command file but i suggest running the script manually to test.

berto547, 2018-08-18:
Wow it seems to be working. Thanks youso much for your help. Appreciate it
very much.
Few questions, I'm sure I will have more down the track.
How often does the page refresh is it with every new message.
How do I get the icons. do I download from link available icons, if so
where do I put the file  or are they in the system.
Once again thanks.

https://i.imgur.com/IarteqL.png

http://192.168.0.10:3000/api/messages

DanrwAU, 2018-08-18:
The website uses a websocket to refresh every time a message is submitted. 

Icons are from the Font Awesome library.
Basically go here, https://fontawesome.com/icons?from=io find the icon you like, then type the name into the box on the alias page. 

See screenshot here for an example of using the icon 'Wifi' 
https://i.imgur.com/e3ptB4A.png

berto547, 2018-08-18:
Thats great, it's working good. With the Psclient every time a page comes pdw to Pagermon the computer pauses and you cant do anything for the split second, is that normal. 
Also able to use pushover for a message word filter instead of a Capcode filter.
Thanks again

DanrwAU, 2018-08-18:
Strange ... Try using the -NoConsole switch in your parameters 

Word filters don't exist for pushover notifications. That is something I intended to look at in https://github.com/davidmckenzie/pagermon/pull/86 and.completely forgot.

Maybe open an issue for it

berto547, 2018-08-18:
Sorry to be a pain, can you be more specific on where to add -NoConsole. Do
you mean in the client file.

I might add it is issues. Really great feature in PDW that I used to get
emails to pushover in regards to Large scale incidents. PagerMon interface
is so much more user friendly.
Cheers

DanrwAU, 2018-08-19:
Sorry NoConsole isn't a valid switch i was thinking of a module i use at work. 

How are you calling the powershell file?

The way i did it when i ran PDW was like this:
https://i.imgur.com/qF3hF0J.png

Command File: C:\Windows\system32\WindowsPowerShell\v1.0\powershell.exe
Arguments: -windowstyle hidden -noprofile -executionpolicy bypass -File C:\Pager.ps1 "%1" "%7"

berto547, 2018-08-19:
I changed it to your settings and still the same issues. powershell opens
at bottom of windows bar for a second every time i get a page which is
alot. Not sure what i have done wrong.

DanrwAU, 2018-08-19:
Powershell will always open, it shouldn't steal focus or interrupt you this way but unfortunately there isn't much else you can do without getting really complicated. 

I ended up switching to a raspberry pi running multimon at each of the sites i'm capturing from.

berto547, 2018-08-19:
Think I might invest in a pi for this long term. I have pdw setup with 2
applications and use sdr to pick up 2 pager frequencies. Would this be
possible with multimon on the pi.
How do you have your pi setup.

davidmckenzie, 2018-08-19:
By default it's not possible with rtl_fm and multimon to track two frequencies with a single dongle - the rtl_fm package can only output one channel. It's trivial with two dongles, but if you wanted to stick with only one dongle then you'd have to look at some kind of intermediary to process the channels and send them to separate pipes. I have a vague recollection that [tsl-sdr](https://github.com/pvachon/tsl-sdr) was something that could potentially do this - no idea how well it plays with multimon though.

berto547, 2018-08-20:
Since I am having luck with it on Windows, I would like to get the power shell not interrupting the pages as the are piped. I have been looking online and it is a common issue. forums are saying to add 
-NoNewWindow just not sure where to add it. Would you know where to put the code.
Cheers

DanrwAU, 2018-08-20:
-NoNewWindow would only work if you were running a powershell window and pasting into it. With the way PDW is calling the script it needs to call the EXE Each time.

When i was running this way, the powershell window would flash on the screen for a second or 3 then disappear. 
Is this not what is happening for you? 

A screenshot of whats occuring might help us help you as well :)

berto547, 2018-08-20:
Thats what is happening except every time it flashes on the taskbar I can't type anything at the same time. Eg if I was typing a document in word every few seconds it will cut out letters whenever powershell process a new page. it's like it pauses for that 1-3 seconds and skips letters.

DanrwAU, 2018-08-20:
Ah ok so powershell is stealing focus... 

The only way i can think of to work around that easily would be to run PDW on a separate windows user account then use switch user to go back to your main one. 

I always ran PDW on a separate crappy laptop hiding in the corner so never had that issue.

berto547, 2018-08-20:
Just reated a new account and have PDW and SRD running pages on the new account. down't seem to be working. Is there anything else I need to do with psclient or anything. Thanks

DanrwAU, 2018-08-20:
I haven't had a chance to test this out myself as yet. 

I'll give it a go tonight and see if it works for me.

DanrwAU, 2018-08-21:
Tried this myself, your right SDR# doesn't like being switched out. 
Easy solutions have now escaped me. 

Got an old spare laptop? ;-)

closed 2018-10-11 by marshyonline