pagermon/pagermon #91: 0.1.5->0.1.6 Dbase failures

kiwiscout, 2018-08-30:
I'm likely alone. And couldn't work out why, other than viewing the dbase to not it not being updated with the push fields..

But on upgrade, aliases weren't able to be checked before display - so showed ignored and all others with active alias as 'unknown'. And when attempting to adjust/save aliases it would give an unknown/invalid error

Ended up forcing head to previous pull and merging back to 0.1.5 seemingly OK.. *touch wood till I see a known address* 

Ideas on why it missed the dbase change and how to manually tweak? I don't use that function but the others added looked to be a good move forward.

DanrwAU, 2018-08-30:
The database isn't automatically upgraded unfortunately. 

It's a fairly easy process, you need to run the following commands

ALTER TABLE capcodes ADD COLUMN pushgroup TEXT;
ALTER TABLE capcodes ADD COLUMN pushsound TEXT;

kiwiscout, 2018-08-30:
That'd to it errrytime :P 

Assuming the issue was the back end referencing incorrect table fields so failing

closed 2018-10-06 by davidmckenzie