pagermon/pagermon #94: Pushover causing application crash

DanrwAU, 2018-09-18:
Noticed this in my logs tonight - Seems to be causing the app to crash and restart, however the notification is sent. 

May be benign but not sure. 

```
2|pagermon |     at IncomingMessage.<anonymous> (/home/user/pagermon/server/node_modules/pushover-notifications/lib/pushover.js:167:12)
2|pagermon |     at emitNone (events.js:111:20)
2|pagermon |     at IncomingMessage.emit (events.js:208:7)
2|pagermon |     at endReadableNT (_stream_readable.js:1064:12)
2|pagermon |     at _combinedTickCallback (internal/process/next_tick.js:138:11)
2|pagermon |     at process._tickDomainCallback (internal/process/next_tick.js:218:9)
2|pagermon | 2018-09-18 06:45 +00:00: (node:3588) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 echo listeners added. Use emitter.setMaxListeners() to increase limit
2|pagermon | Error: user identifier is not a valid user, group, or subscribed user key
2|pagermon |     at Pushover.errors (/home/user/pagermon/server/node_modules/pushover-notifications/lib/pushover.js:84:13)
2|pagermon |     at IncomingMessage.<anonymous> (/home/daniel/pagermon/server/node_modules/pushover-notifications/lib/pushover.js:167:12)
2|pagermon |     at emitNone (events.js:111:20)
2|pagermon |     at IncomingMessage.emit (events.js:208:7)
2|pagermon |     at endReadableNT (_stream_readable.js:1064:12)
2|pagermon |     at _combinedTickCallback (internal/process/next_tick.js:138:11)
2|pagermon |     at process._tickDomainCallback (internal/process/next_tick.js:218:9)
```
  
```

marshyonline, 2018-09-19:
Things to check

- API Keys is OK

- USER/Group Key is ok

This error would tell me its getting a 400 response. 
Can add some debug code to print the response form PushOver

marshyonline, 2018-09-19:
https://github.com/davidmckenzie/pagermon/blob/ee1c3c750925b0761608c82c79bcaeeedc39c874/server/routes/api.js#L784

Change this line to console.log(err);
Should give you an error and not boot you out

DanrwAU, 2018-09-19:
I'll throw it in shortly and see if i can make it happen, i tried to do so later on last night and couldn't get it to happen again.

marshyonline, 2018-09-19:
Ill drop it a PR tonight to stop it killing the app when it hits ann error

davidmckenzie, 2018-09-19:
Was it happening after the change to push settings per alias, or only started more recently?

DanrwAU, 2018-09-19:
Great question, i'm not sure. 

I'll pull some logs and take a looksie soon.

marshyonline, 2018-09-19:
I don't recall ever seeing it in either version :(

DanrwAU, 2018-09-19:
Looks like the issue did occur again today -
` 2|pagermon | 2018-09-19 02:01 +00:00: (node:91197) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 echo listeners added. Use emitter.setMaxListeners() to increase limit`

Thats the only entry in my node-app.stderr.log file. 
Doesn't correspond to a pushover notification this time.

DanrwAU, 2018-09-19:
Just happened again simply from loading the page - Possibly the Sockets aren't being closed correctly after our api security changes?

marshyonline, 2018-09-19:
This is looking like the case, I will debug shortly

marshyonline, 2018-09-19:
Pull this branch and let me know how it goes?
I cant replicate the issue :(

https://github.com/marshyonline/pagermon/tree/EventFixes

DanrwAU, 2018-09-19:
I can't seem to either now, without pulling your branch down. 

I'll do some testing and see if i can figure out how to reproduce then pull it down. 😭

marshyonline, 2018-09-20:
see #95 re: sending errors to the error log and now killing the app on an error in pushover

davidmckenzie, 2018-09-20:
Merged that in, lets see if we get any more descriptive errors now. Wonder if it was a transient issue on the pushover side

DanrwAU, 2018-09-20:
I've pulled it onto the instance that was having the issue, i'll keep an eye out for errors. 

Didn't get a chance yesterday, was busy stealing free stuff from Vantage 😛

DanrwAU, 2018-09-21:
Got another one! 

2|pagermon | 2018-09-21 02:13 +00:00: (node:65698) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 echo listeners added. Use emitter.setMaxListeners() to increase limit

Nothing in the logs surrounding it 

2018-09-21 01:44 +00:00: 1.2.3.4 - - [21/Sep/2018:01:44:15 +0000] "POST /api/messages HTTP/1.0" 200 5 "-" "-"
2018-09-21 01:46 +00:00: 1.2.3.4 - - [21/Sep/2018:01:46:43 +0000] "POST /api/messages HTTP/1.0" 200 5 "-" "-"
2018-09-21 01:47 +00:00: 1.2.3.4 - - [21/Sep/2018:01:47:37 +0000] "POST /api/messages HTTP/1.0" 200 5 "-" "-"
2018-09-21 01:52 +00:00: 1.2.3.4 - - [21/Sep/2018:01:52:27 +0000] "POST /api/messages HTTP/1.0" 200 5 "-" "-"
2018-09-21 01:57 +00:00: 1.2.3.4 - - [21/Sep/2018:01:57:10 +0000] "POST /api/messages HTTP/1.0" 200 5 "-" "-"
2018-09-21 02:14 +00:00: 1.2.3.4 - - [21/Sep/2018:02:14:26 +0000] "POST /api/messages HTTP/1.0" 200 5 "-" "-"
2018-09-21 02:34 +00:00: 1.2.3.4 - - [21/Sep/2018:02:34:01 +0000] "POST /api/messages HTTP/1.0" 200 5 "-" "-"

marshyonline, 2018-09-21:
Can you check to make sure that user/group key is correct?

Can you also please add `io.sockets.setMaxListeners(20);` around line 103 of app.js?

I'm wondering if the new admin socket has put it on its limit

DanrwAU, 2018-09-21:
Figured it out! 

I have an alias with 'Enable sending via Pushover' Enabled without a key! 

We should probably put some checks around that for stupid people like myself :P 

The Socket error seems to be independant of the pushover error - i just happened to get them at the same time the first time i saw it

marshyonline, 2018-09-21:
Well, that's good to know :rofl: 
Checking should be easy to add - Just make it a required field on the client side.

As above, give that a crack and see if it spits the dummy again

marshyonline, 2018-09-21:
Looks like its the actual pushover Libary causing it to throw the application out the window 
https://github.com/qbit/node-pushover/blob/4fa9d9a067acd2e2756428b9bd09901f9333a933/lib/pushover.js#L84

DanrwAU, 2018-09-23:
I haven't seen the socket error since setting the max listeners to 20. I'd guess the Admin socket pushed it a little high 😄

marshyonline, 2018-09-23:
Beautiful 💕 
Thats what I like to hear 

I shall push it tonight 😄

DanrwAU, 2018-09-23:
I'll fix up the Commit for the pushover side tomorrow if i get time. I had a go at the GUI side but couldn't get the Javascript to work properly 😭

marshyonline, 2018-09-23:
#96 is open :D

DanrwAU, 2018-09-26:
https://github.com/davidmckenzie/pagermon/pull/97 Should stop pushover crashing the application when an User/Group key isn't set.

I'd like to do some validation in the GUI on these fields but Javascript isn't my strong suit 😞

marshyonline, 2018-10-11:
This has been resolved :D

closed 2018-10-11 by marshyonline