pagermon/pagermon #99: Searching messages - High Memory Usage

DanrwAU, 2018-09-30:
Another one to add to the list.

Searching messages seems to cause significant memory usage server side with large databases. 

Performing a search for 2 words consistently triggers the application to run away above 1GB of ram - at which point PM2 kills the process and restarts. 

Database is currently 25MB (ish) with 100k messages.

davidmckenzie, 2018-09-30:
This old chestnut... really need to switch to better DB engine.

In logs should be some timers for each search query - can you grab those?

Does it make a difference if you're auth'd or not? Wonder if it's falling over on line 378 when hidecapcode is enabled and unauth'd

```
search.addIndex('address');
```

DanrwAU, 2018-10-01:
This one worked - not signed in. 
[image]

Also worked - logged in 
[image]

I need to dig back and find the query i was using last night to crash it :)

DanrwAU, 2018-10-01:
This one pushed the memory right up to around 1GB - fell just short of restarting the app... 

[image]

marshyonline, 2018-10-01:
`DB Stats:
Total Messages: 303758
DB Size: 42M

Home Page Load:
2018-10-01 11:33 +10:00: init: 128.484ms
2018-10-01 11:33 +10:00: sql: 3.937ms
2018-10-01 11:33 +10:00: send: 2.362ms

Search FireCall(Once)
2018-10-01 11:34 +10:00: init: 0.294ms
2018-10-01 11:34 +10:00: sql: 4926.012ms
2018-10-01 11:34 +10:00: search: 4.388ms
2018-10-01 11:35 +10:00: searchFullText: 23070.355ms
2018-10-01 11:35 +10:00: sort: 25.797ms
2018-10-01 11:35 +10:00: initEnd: 0.066ms

Search FireCall(3 tabs):
2018-10-01 11:36 +10:00: sql: 4883.396ms
2018-10-01 11:36 +10:00: init: 0.390ms
2018-10-01 11:36 +10:00: sql: 4700.580ms
2018-10-01 11:36 +10:00: Listening on port 3000
2018-10-01 11:36 +10:00: search: 4.191ms
2018-10-01 11:36 +10:00: searchFullText: 19013.560ms
2018-10-01 11:36 +10:00: sort: 23.855ms
2018-10-01 11:36 +10:00: initEnd: 0.128ms
Gracefully shutting down from SIGINT (Ctrl-C) :sad:

Logged out with HideCapcode on:
2018-10-01 11:43 +10:00: ***IP*** - - [01/Oct/2018:01:43:38 +0000] "GET /?q=FireCall HTTP/1.1" 304 - "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/69.0.3497.100 Safari/537.36"
2018-10-01 11:43 +10:00: ***IP*** - - [01/Oct/2018:01:43:38 +0000] "GET /stylesheets/style.css HTTP/1.1" 304 - "https://SNIP/?q=FireCall" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/69.0.3497.100 Safari/537.36"
2018-10-01 11:43 +10:00: init: 0.412ms
2018-10-01 11:43 +10:00: sql: 5621.985ms
2018-10-01 11:43 +10:00: search: 5.018ms
2018-10-01 11:44 +10:00: searchFullText: 21258.989ms
2018-10-01 11:44 +10:00: sort: 21.985ms
2018-10-01 11:44 +10:00: initEnd: 0.067ms
2018-10-01 11:44 +10:00: Listening on port 3000
2018-10-01 11:44 +10:00: 180.181.100.84 - - [01/Oct/2018:01:44:05 +0000] "GET /api/messageSearch?limit=&page=1&q=FireCall HTTP/1.1" 200 - "https://SNIP/?q=FireCall" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/69.0.3497.100 Safari/537.36"
`

marshyonline, 2018-10-01:
Looks like there is an issue just in general  :( 
It died regardless if the user was logged in or not

davidmckenzie, 2018-10-01:
I'm still on an ancient version... that searchFullText bit hasn't changed much since then though

Single word:
```
3|pagermon | 2018-10-01 06:02 +00:00: init: 0.114ms
3|pagermon | 2018-10-01 06:02 +00:00: sql: 431.599ms
3|pagermon | 2018-10-01 06:02 +00:00: search: 1.143ms
3|pagermon | 2018-10-01 06:02 +00:00: searchFullText: 2646.166ms
3|pagermon | 2018-10-01 06:02 +00:00: sort: 7.150ms
3|pagermon | 2018-10-01 06:02 +00:00: initEnd: 0.028ms
```

Two words:
```
2018-10-01 06:03 +00:00: init: 0.231ms
2018-10-01 06:03 +00:00: sql: 719.594ms
2018-10-01 06:03 +00:00: search: 1.060ms
2018-10-01 06:03 +00:00: searchFullText: 4377.049ms
2018-10-01 06:03 +00:00: sort: 0.362ms
2018-10-01 06:03 +00:00: initEnd: 0.019ms
```

My messages.db is only 4.4M, thinking this might just be us hitting critical mass on DB size again. I'll pull down the latest and see if there's any difference.

davidmckenzie, 2018-10-01:
Yep pulled down latest and still the same results, so definitely hitting limitations on DB size again.

Wonder if we could use this instead, will have to have a look: http://www.sqlitetutorial.net/sqlite-full-text-search/

davidmckenzie, 2018-10-01:
Can you guys run this on your sqlite installs, fts is compiled by default on the version I'm using, not sure about other distros:

```
sqlite> PRAGMA compile_options;
ENABLE_COLUMN_METADATA
ENABLE_FTS3
ENABLE_RTREE
ENABLE_UNLOCK_NOTIFY
ENABLE_UPDATE_DELETE_LIMIT
MAX_SCHEMA_RETRY=25
OMIT_LOOKASIDE
SECURE_DELETE
SOUNDEX
SYSTEM_MALLOC
TEMP_STORE=1
THREADSAFE=1
```

^ is on ubuntu 14

marshyonline, 2018-10-01:
I run in CentOS 7 
I can give it a crack later tonight 😃

DanrwAU, 2018-10-01:
COMPILER=gcc-6.4.0 20170704
DEFAULT_SYNCHRONOUS=2
DEFAULT_WAL_SYNCHRONOUS=2
ENABLE_COLUMN_METADATA
ENABLE_DBSTAT_VTAB
ENABLE_FTS3
ENABLE_FTS3_PARENTHESIS
ENABLE_FTS4
ENABLE_FTS5
ENABLE_JSON1
ENABLE_LOAD_EXTENSION
ENABLE_RTREE
ENABLE_UNLOCK_NOTIFY
ENABLE_UPDATE_DELETE_LIMIT
HAVE_ISNAN
LIKE_DOESNT_MATCH_BLOBS
MAX_SCHEMA_RETRY=25
OMIT_LOOKASIDE
SECURE_DELETE
SOUNDEX
SYSTEM_MALLOC
THREADSAFE=1

marshyonline, 2018-10-01:
CentOS 7.5
```
sqlite> PRAGMA compile_options;
DISABLE_DIRSYNC
ENABLE_COLUMN_METADATA
ENABLE_FTS3
ENABLE_RTREE
ENABLE_UNLOCK_NOTIFY
SECURE_DELETE
TEMP_STORE=1
THREADSAFE=1
```

davidmckenzie, 2018-10-04:
So looks like FTS3 is going to be supported on pretty much every modern distro. Shame FTS4/5 isn't more prevalent. I'll have a play with this over the weekend/next week as I've got some free time coming up.

If I'm feeling particularly ambitious I might take another look at porting to a different backend DB stack ;)

davidmckenzie, 2018-10-04:
I'll also take another look at the indexes we have on the DB - may need to revise them with the query changes for emails and pushover etc.

davidmckenzie, 2018-10-06:
PR submitted #102

closed 2018-10-06 by davidmckenzie