pagermon/pagermon #108: FLEX multipart Messages

Shaggs, 2018-10-22:
Hi All,
    Im trying to contribute to this project but as I'm from a python programming no JS I'm having some issues joining the messages together

Multimon now supports fragmented messages by using a letter being K (complete page) F (fragmented message) and C (continued) . This can be found in the FLEX string here being represented by the *

`1600/2/ * /A`  

what my aim was in using reader.js to add these stings together and then upload.

an example sting as follows
`2018-10-22 02:06:45 1600/2/F/A [001234567] ALN TH`
`2018-10-22 02:06:45 1600/2/C/A [001234567] ALN IS IS A TEST PAGE`
`2018-10-22 02:07:45 1600/2/K/A [001234567] ALN THIS IS A SECOND TEST PAGE`

i can see in reader.js regex is used to match the word FLEX. Is it possible to add another regex to find the find the `F` and store the message and capcode. Then if the next  page contains the `C` compare the capcodes and if they match join up the messages and upload. Then the other strings can be cleared.

Hope this makes sense.

happy to test and supply more examples or can save audio if required

Shaggs

davidmckenzie, 2018-10-22:
That's an interesting one - do the codes on either side of it change or could you look for just "/2/F/A"? I suppose you could look for a string matching \d{4}/[FCK]/. and then use the middle bit.

The logic might need a bit more thinking - if it's fragmented multiple times does it go F -> C -> C, or F -> F -> C? Can other messages jump the queue (F -> K -> C), or could you get two fragmented messages simultaneously (F1 -> F2 -> C1 -> C2)?

How would we handle an F that's never followed by a C?

Shaggs, 2018-10-22:
> That's an interesting one - do the codes on either side of it change or could you look for just "/2/F/A"? I suppose you could look for a string matching \d{4}/[FCK]/. and then use the middle bit. [...]

From my experience the code either side does not change. The messages only fragment once from what i can see. 

My thoughts were if there is an F and the next is not a C then just dispose of the page. But guess in server settings could add a switch for it. 

As for can message jump I've never looked (research underway) 

I have never seen fragmented messages simultaneously. Only ever one at a time. 

Best thing would be add the change as a branch and I can test it out and post results after 24-48 hrs

Shaggs, 2018-10-22:
Can confirm messages can not jump but what can happen is you only get one part of the fragment and not the other depending on reception. Thats why i think a switch would be good.

The way i handle this in python is by checking if there is an F if there is keep it until the next message comes in and if it is a C and the capcodes match then joining F to C then print to screen.

>> could you get two fragmented messages simultaneously (F1 -> F2 -> C1 -> C2)?

No as multimon has not got this function and don't think it can happen in FLEX

davidmckenzie, 2018-10-22:
Thanks, starting to get a better picture of how this would look. I don't have any FLEX systems around here, but just looking through previous issues (#28) I can see that user's format is "1600/2/A" for that bit of status code stuff.

I was going to ask if that's system specific then did a little more digging and found that this is relatively recent in multimon:

```
*  Version 0.8.7v (11 APR 2018)
 *  Modification (to this file) made by Bruce Quinton (zanoroy@gmail.com) and Rob0101 (as seen on github: https://github.com/rob0101)
 *     - Issue *#95 created by rob0101: '-a FLEX dropping first character of some message on regular basis'
 *     - Implemented Rob0101's suggestion of K, F and C flags to indicate the message fragmentation: 
 *         'K' message is complete and O'K' to display to the world.
 *         'F' message is a 'F'ragment and needs a 'C'ontinuation message to complete it. Message = Fragment + Continuation
 *         'C' message is a 'C'ontinuation of another fragmented message
```

So we'd need to look at backwards compatibility but otherwise shouldn't need to worry about differences in systems. :)

davidmckenzie, 2018-10-22:
The other bits there are: `flex->Sync.baud, flex->Sync.levels, frag_flag, PhaseNo` - frag_flag is the fragmentation flag. It and PhaseNo are strings, while the first two are integers. Should be simple enough to regex something to pull out the frag_flag.

Shaggs, 2018-10-27:
ill have a crack at this later today and I'm sorry if code is bad but hey I'm a python user

Shaggs, 2018-10-27:
fixed will pull request #110

Shaggs, 2018-11-03:
Testing now underway with latest update

Shaggs, 2018-11-20:
Sorry @davidmckenzie.  Remember how you asked if pager messages can jump?  The answer is now yes. 

With an update to FLEX here is South Australia, a fragmented message can now have another page inbetween. So I'm sorry but back to the drawing board on this one. 

I'm struggling to find a way in python to handle this so wondered if you had an idea in node and I can make the same in python.

davidmckenzie, 2018-11-20:
It *should* be ok with fragmented messages having another in between them, as long as that other message isn't also fragmented.

If the other message is fragmented not sure how you'd match them up. I guess matching on address would be about as foolproof as it gets.

Could maybe set frag to be an object, something like this:

```
frag[address] = line.match(/FLEX:.*?\[\d*\] ... (.*?)$/)[1].trim();
```

Then instead of just adding `frag`, you:
```
trimMessage = frag[address]+message;
delete frag[address]
```

Shaggs, 2018-11-20:
We could give that a go... I can send you an audio file if needed so you can look at it closer if it helps. 

Tomorrow I'll fire up the rtl again.  Currently I've been piping from pdw to pagermon via python.

davidmckenzie, 2018-12-06:
Just a heads up, haven't forgotten about this, will crank something out soon, and hopefully knock out #76 at the same time :)

PR #173 (2019-01-08): Flex frag + retries
Resolves #108 and #76 

/cc @Shaggs and @112bzo as FLEX users

This moves the FLEX fragmented messages into an object with the address as key - fragmented messages that arrive with other messages in between should now properly join. The only real edge case I can think of is if in poor reception and you receive an F message, but never its corresponding C message, then you might one day receive a randomly joined message. Would be rare enough not to worry about, IMO.

The other change in this is the retries that @s3m1s0n1c requested - messages will retry sending on failure up  [... cut]
closed 2019-01-08 by davidmckenzie