pagermon/pagermon #112: Telegram Push Notifications

s3m1s0n1c, 2018-11-01:
**Enhancement**

Integrate Push Notifications into Telegram so users can be notified of their brigade's calls.

This will basically work the same as Pushover Notifications..

Set in each Alias which Telegram Channel/Group you wish to send notification too.
Set API/Bot Key in Settings.

Telegram Documentation: https://core.telegram.org/

DanrwAU, 2018-11-01:
I've seen a fork with this implemented already, may just need to be cleaned up and implemented cleanly with the existing notification stuff. 

I might take a looksie at this one when i get some time.

davidmckenzie, 2018-11-01:
That's a good idea, I likey. I've done some telegram integration work before, but only with userbots. I'll also have a look at some point if I get the time :)

DanrwAU, 2018-11-01:
Test this one out ... Not 100% it is bug free yet and not really happy with the message format but i couldn't come up with anything better with a 2yo climbing all over my lap 😛 
https://github.com/davidmckenzie/pagermon/pull/113

I reused some of the code from this fork https://github.com/e-bits/pagermon/tree/feat/telegram where he added telegram support, but set it up the same way as pushover. 

If you want to write some documentation on how to get a API Key and Chat ID @s3m1s0n1c  feel free, that was a damn pain! :wink:

s3m1s0n1c, 2018-11-01:
Wow.. That was quick... Thanks Dan

Will give it a whirl and see how it goes..  Yeah i can imagine having a 2yr old climbing around would make it more difficult.. Don't have that issue YET.

Yeah getting the API Key and Chat ID is a pain in the butt..  I will write some instructions on how to get get the info..

Thanks..

davidmckenzie, 2018-11-04:
Solved in #113 thanks guys!

closed 2018-11-04 by davidmckenzie