pagermon/pagermon #115: Pushover Crash - XML Encoding

DanrwAU, 2018-11-02:
Got a crash tonight with this error 
```
1|pagermon | Error: XML parameter input not accepted, please use standard percent-encoding (application/x-www-form-urlencoded) or JSON (application/json) encoding
1|pagermon |     at Pushover.errors (/srv/pagermon/server/node_modules/pushover-notifications/lib/pushover.js:141:13)
1|pagermon |     at IncomingMessage.<anonymous> (/srv/pagermon/server/node_modules/pushover-notifications/lib/pushover.js:241:12)
1|pagermon |     at emitNone (events.js:111:20)
1|pagermon |     at IncomingMessage.emit (events.js:208:7)
1|pagermon |     at endReadableNT (_stream_readable.js:1064:12)
1|pagermon |     at args.(anonymous function) (/usr/local/lib/node_modules/pm2/node_modules/event-loop-inspector/index.js:138:29)
1|pagermon |     at _combinedTickCallback (internal/process/next_tick.js:138:11)
1|pagermon |     at process._tickDomainCallback (internal/process/next_tick.js:218:9)
```
Nothing special about the message, pushing it manually from the powershell client worked fine. 

Not one i've seen before ☹️

davidmckenzie, 2018-11-02:
What was the message, if you don't mind sharing? The error makes me think of URI encoding of special chars.

DanrwAU, 2018-11-02:
2018-11-02 20:54:42: 0572008: @@ALERT F181101061 ROCK4 RESCC3 * CAR ACCIDENT - POSS PERSON TRAPPED MELTON HWY MELTON M 338 D8 (916269) CCARO CMTON CROCK MTONR [CARO]

davidmckenzie, 2018-11-02:
Thanks, yeah that's super weird, can't see why it would do that. None of the special chars are *that* special, and looking at the error again I think it's saying that it interpreted the input as XML... 

Reminds me of in another app where I had to send unparsed output from one API to another, when the source API returned an error the "output" would be a pretty HTML formatted error message, which then got sent to the other API as is, who would think it's XML.

Can't see how that could happen here though, unless the alias in question got dirty data stored in the DB somehow? Maybe worth taking a look at that message's row in the DB, and the alias's row, see if anything odd there.

Only other idea is if something changed in pushover API, but then it'd be erring out on every message

DanrwAU, 2018-11-02:
Yep, it's got me flummoxed as well, the row in the database looks normal. 

Might have just been a one of glitch, i'll keep an eye on my error logs and see if it comes back 🤷‍♂️

marshyonline, 2018-11-05:
i think this may have just been a little blip on the Pushover side of things
Their API hasn't changed from what i can see and i use it for a few things.

DanrwAU, 2018-11-05:
Agreed, I've checked my logs and hasn't reoccured. I reckon we close this one. 😄

marshyonline, 2018-11-06:
Ive got some monitoring setup on logs to try  and catch this stuff in the future
Feel free to re open with we can catch it again 😄

closed 2018-11-06 by marshyonline