pagermon/pagermon #116: Secure Mode - Searching Fails

DanrwAU, 2018-11-02:
Another one from me, i should stop reading logs.

Looks like searching is broken when secure mode is enabled. 
Searches return no results and the following is thrown at the console.
```
0|pagermon |     at /srv/pagermon/server/node_modules/express/lib/router/index.js:281:22
0|pagermon |     at param (/srv/pagermon/server/node_modules/express/lib/router/index.js:354:14)
0|pagermon |     at param (/srv/pagermon/server/node_modules/express/lib/router/index.js:365:14)
0|pagermon |     at Function.process_params (/srv/pagermon/server/node_modules/express/lib/router/index.js:410:3)
0|pagermon |     at next (/srv/pagermon/server/node_modules/express/lib/router/index.js:275:10)
0|pagermon |     at /srv/pagermon/server/routes/api.js:25:3
0|pagermon |     at Layer.handle [as handle_request] (/srv/pagermon/server/node_modules/express/lib/router/layer.js:95:5)
0|pagermon |     at trim_prefix (/srv/pagermon/server/node_modules/express/lib/router/index.js:317:13)
0|pagermon |     at /srv/pagermon/server/node_modules/express/lib/router/index.js:284:7
0|pagermon |     at Function.process_params (/srv/pagermon/server/node_modules/express/lib/router/index.js:335:12)
0|pagermon |     at next (/srv/pagermon/server/node_modules/express/lib/router/index.js:275:10)
0|pagermon |     at urlencodedParser (/srv/pagermon/server/node_modules/body-parser/lib/types/urlencoded.js:91:7)
0|pagermon |   name: 'AuthenticationError',
0|pagermon |   message: 'Unauthorized',
0|pagermon |   status: 401 }
```
This one is above my paygrade unfortunately 😞

marshyonline, 2018-11-02:
Interesting
You will always see one auth failed there because it checks API key and then isLoggedIn

it looks like its not passing isLoggedIn correctly to the get and failing. 
Ill look into this tonight 💃

DanrwAU, 2018-11-04:
This seems to fix the issue, however now triggers the auth failure for other actions (loading the homepage, loading an alias settings screen)

The page still loads correctly however not sure that is desired behavior.

```
pagermon > 2018-11-04 13:48 +00:00: 127.0.0.1 - - [04/Nov/2018:13:48:40 +0000] "GET /stylesheets/style.css HTTP/1.0" 304 - "https://dev01.home.local/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/70.0.3538.77 Safari/537.36" 
pagermon > 2018-11-04 13:48 +00:00: { AuthenticationError: Unauthorized                                                                                           x
pagermon >     at allFailed (/srv/pagermon/server/node_modules/passport/lib/middleware/authenticate.js:159:21)                                                    x
pagermon >     at attempt (/srv/pagermon/server/node_modules/passport/lib/middleware/authenticate.js:167:28)                                                      x
pagermon >     at Strategy.strategy.fail (/srv/pagermon/server/node_modules/passport/lib/middleware/authenticate.js:284:9)                                        x
pagermon >     at Strategy.authenticate (/srv/pagermon/server/node_modules/passport-localapikey-update/lib/passport-localapikey/strategy.js:76:17)                x
pagermon >     at attempt (/srv/pagermon/server/node_modules/passport/lib/middleware/authenticate.js:348:16)                                                      x
pagermon >     at authenticate (/srv/pagermon/server/node_modules/passport/lib/middleware/authenticate.js:349:7)                                                  x
pagermon >     at Layer.handle [as handle_request] (/srv/pagermon/server/node_modules/express/lib/router/layer.js:95:5)                                           x
pagermon >     at next (/srv/pagermon/server/node_modules/express/lib/router/route.js:137:13)                                                                     x
pagermon >     at Route.dispatch (/srv/pagermon/server/node_modules/express/lib/router/route.js:112:3)                                                            x
pagermon >     at Layer.handle [as handle_request] (/srv/pagermon/server/node_modules/express/lib/router/layer.js:95:5)                                           x
pagermon >   name: 'AuthenticationError',                                                                                                                         x
pagermon >   message: 'Unauthorized',                                                                                                                             x
pagermon >   status: 401 }                                                                                                                                        x
pagermon > 2018-11-04 13:48 +00:00: init: 48.399ms                                                                                                                x
pagermon > 2018-11-04 13:48 +00:00: sql: 1.945ms                                                                                                                  x
pagermon > 2018-11-04 13:48 +00:00: send: 1.193ms                                                                                                                 x
pagermon > 2018-11-04 13:48 +00:00: 127.0.0.1 - - [04/Nov/2018:13:48:40 +0000] "GET /api/messages?limit=&page=1 HTTP/1.0" 304 - "https://dev01.home.local/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/70.0.3538.77 Safari/537.36"                                           x
pagermon > 2018-11-04 13:48 +00:00: 127.0.0.1 - - [04/Nov/2018:13:48:41 +0000] "GET /favicon-32x32.png?v=yyx75bRMkG HTTP/1.0" 304 - "https://dev01.home.local/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/70.0.3538.77 Safari/537.36"                                       x
pagermon > 2018-11-04 13:48 +00:00: 127.0.0.1 - - [04/Nov/2018:13:48:41 +0000] "GET /manifest.json?v=yyx75bRMkG HTTP/1.0" 304 - "https://dev01.home.local/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/70.0.3538.77 Safari/537.36"
pagermon > 2018-11-04 13:48 +00:00: 127.0.0.1 - - [04/Nov/2018:13:48:41 +0000] "GET /favicon-16x16.png?v=yyx75bRMkG HTTP/1.0" 304 - "https://dev01.home.local/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/70.0.3538.77 Safari/537.36"       
```

marshyonline, 2018-11-04:
Interesting, i didnt see this in dev.
I shall look over it and see whats going on.

The auth login is confusing as shit right now 😆

marshyonline, 2018-11-04:
Ah ok i see it
anything below this will throw an auth error
`// secure all API's if API Security is enabled
if (apiSecurity) {
  router.all('*',
    passport.authenticate('localapikey', { session: false, failWithError: true }),
    function(req, res, next) {
      next();
    },
    function(err, req, res, next) {
      console.error(err);
      isLoggedIn(req, res, next);
    }
  );
};
`
This is because it is looking for an API key in the request before trying the isLoggedIn function.
If you hit /api/messages you should get the same, the error should print once if you are logged in and twice if you are not.

I'm going to look at cleaning this up, but at the moment there is no way around this

davidmckenzie, 2018-11-05:
Solved in #118

closed 2018-11-05 by davidmckenzie