pagermon/pagermon #119: Timestamp Formating for both US and AU

s3m1s0n1c, 2018-11-05:
Would be nice to be able to change the date and choose between US or AU..

Obviously, once this is set it cannot be changed as the DB will be in on or the other format.

The way around this is if you save the time as a unixtimestamp in the DB and then you can convert it to whatever format you want.

You prob would have to add some sort of conversion for those ppl already with DB's tho

s3m1s0n1c, 2018-11-05:
I'm gonna give this one a go and see if i can do it :)

davidmckenzie, 2018-11-05:
The time should display in the browser's localtz, I opted for the international date format of YYYY-MM-DD cos it's the most universally known, but you *could* output in the browsers default format.

See: https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Date/toLocaleDateString

Alternatively, could leave the datetime format as is, and then add a tooltip when hovering over the date/time that shows the full locale string