pagermon/pagermon #122: Documentation for Pagermon

s3m1s0n1c, 2018-11-06:
Have started better documentation for Pagermon.

Below is a link to my pagermon wiki that is currently in DEV state
https://github.com/s3m1s0n1c/pagermon/wiki

Feel free to comment or make suggestions.

I think I'll leave pics out of this one.. takes up too much space and takes too much time for now..

marshyonline, 2018-11-06:
Its looking good so far 😄 
One suggestion i will make, is to make use of pages.

Currently everything is on one page, splitting things up into pages will make nicer on the eyes

s3m1s0n1c, 2018-11-06:
I disagree, it's no different to readme.md
You can use the menu tree on the right to go to any one section and find info..

@davidmckenzie What format would you like this in?

davidmckenzie, 2018-11-06:
I'm mostly indifferent - single page makes sense for a basic readme, but isn't really scalable once you start going into detailed topics. I'm more used to documenting on confluence where you can go middle ground and have fancy expando thingies.

If I was writing it I'd probably split into separate pages to make editing easier, but that's just my personal preference :)

DanrwAU, 2018-11-06:
Looking good so far. 

My personal preference is seperate pages for each block but again that's just my personal preference :-)

I've been meaning to write down my process for spinning up/down instances and setting up NGINX for reverse proxying with certbot for ssl. 

I'll write it down next time i break my test env completely and throw in a pull :smile:

s3m1s0n1c, 2018-11-08:
mmmm guess the tribe has spoken...

I have made updates and added pages..  Let me know if that's the right direction everyone wants.

@DanrwAU  yeah cool if you could write it down and i'll add it.
Going to document Apache proxy as well for those more hardcore and have server and client separate.

Thanks

marshyonline, 2018-11-08:
Lookin really good @s3m1s0n1c !

davidmckenzie, 2018-11-09:
+1, looking good!

112bzo, 2018-11-12:
+1 from me

s3m1s0n1c, 2018-11-15:
Note to self.. Add Signal Flow Diagram to Documentation :)

s3m1s0n1c, 2018-12-18:
I haven't forgotten about this..

Just been busy with life/works.

Thanks

closed 2025-12-09 by s3m1s0n1c