pagermon/pagermon #124: Add documentation for Text Matching

pyrog, 2018-11-08:
Could you provide a screen copy of the settings page to show how text matching could be used ?

How to:
* Replace **RCR** by **Road Crash Rescue** ?
* Highlight **ALERT** or **FIRECALL** ?
* Use **regex** ?

Thank you 😀

DanrwAU, 2018-11-09:
Hi @pyrog 

You can't do any replacing in the current implementation, only labeling on hover.
Highlighting of the word FIRECALL is in the default configuration, i highlight alert in my config as below.

[image]

Regex is a painful beast but very powerful once you know how to use it. 
I use regex for highlighting street directory addresses

See the below regex string and the text it matches. 

[image]
[image]

Did this help at all?

davidmckenzie, 2018-11-09:
Here's my text matching settings if it helps:

[image]

Not using any regex there, but can see highlight vs text replacing.

@s3m1s0n1c is working on some documentation - may be worth adding to the list of things that need words

closed 2018-11-12 by marshyonline