pagermon/pagermon #127: Twitter

112bzo, 2018-11-12:
I see pushover and telegram in version 0.1.9 but is it posseble to make also a auto tweet API ?

s3m1s0n1c, 2018-11-12:
Possible.. Just have to watch to not flood twitter as they have a rate limit for there api
I think its 15 calls per 15 min for memory last time i played with it...

davidmckenzie, 2018-11-12:
I feel like we need a generic plugin framework to handle all these additional things :)

DanrwAU, 2018-11-12:
> I feel like we need a generic plugin framework to handle all these additional things :)I feel like we need a generic plugin framework to handle all these additional things :)

Probably! In the mean time i might have some time to set this up tomorrow, doesn't seem to difficult based on the npm twitter module.

s3m1s0n1c, 2018-11-12:
Add the alias as a #hashtag at the end of message so u can sort by it?

DanrwAU, 2018-11-13:
I've been working for 30 mins or so, however it looks to me like Twitters Developer Applicationy process is quite the PITA, i'm not sure how feasible it would be for end users to go through this process. (including a minimum 300 character answer on how you will be using their API).

I'll try and get it finished (assuming i get approved for a dev account) but it may be something we decide is too cumbersome i suspect.

112bzo, 2018-11-13:
I have now the 0.1.9 beta but when i use Telegramm API my Pi crash. Dont use the API telegramm your Pi wil go offline. Now just re-install all and it works again. 

So i look forwart to the twitter function... Telegramm i dont use anymore.

DanrwAU, 2018-11-13:
Sorry that happened to you but we really need a lot more info than what you've given.

I've been using the plugin all day while building the Twitter version and haven't had an issue on a fresh install.

You say you needed to reinstall your Pi, that isn't something a node app can cause as far as I know, do you have any logs/error messages etc of what happened?

Also what OS, version of Pi etc.

Thanks

DanrwAU, 2018-11-13:
Twitter is done here https://github.com/DanrwAU/pagermon/commit/ac8d21d821cfda9cacbebeea26942c2eb0e8d6de Waiting on Twitter to give me a developer account so i can test the actual posting, everything else looks good.

112bzo, 2018-11-13:
> Sorry that happened to you but we really need a lot more info than what you've given.
>
> I've been using the plugin all day while building the Twitter version and haven't had an issue on a fresh [...]

Ik use the Raspberry Pi 3 B+ and mu SD only contain the Pagermon. What i see is that when i use the key into the admin the node go over kill. 
So its a NODE problem i  think. 
What dor you need to see en how do i get that info so i can share.

DanrwAU, 2018-11-13:
So when you put the Bot API Key Here 
[image]

Node crashes?

We'll need to see the error logs, are you using PM2?

112bzo, 2018-11-13:
Thats true, PM2?

DanrwAU, 2018-11-13:
If you aren't using PM2, the error should be printed to the console when node crashes, we need these error messages.

DanrwAU, 2018-11-16:
Solved with https://github.com/davidmckenzie/pagermon/pull/129

closed 2018-11-16 by DanrwAU