pagermon/pagermon #131: internal server error when editing alias

mitch2k, 2018-11-14:
Hi,

first of all a big thank you for this great piece of software!

I was running PM 0.1.5 on an Raspberry Pi at home where my SDR dongle is also connected. I reinstalled the server at a CentOS on a VM. I've used the version 0.1.9 for the new install. I have moved my messages.db to the new VM. After starting the server I saw that all 'old' messages and aliases were there.

I know there were some DB changes between 0.1.5 and 0.1.9. I inspected the DB and could see the telegram related columns, so the DB upgrade did it's thing.

Everything works perfect, except that I can't do anything with aliases anymore. No editing, removing, adding etc. Each time when I save I get "Error saving alias: [object Object]".

In the console I can see:
Failed to load resource: the server responded with a status of 500 () /api/capcodes/1610

Object data:

> Object
> config:
> data: l
> $promise: f {$$state: {…}}
> $resolved: true
> address: "01005__  "
> agency: "BRW"
> alias: "Algemeen"
> color: "#FF0000"
> icon: "fire"
> id: 1610
> ignore: 0
> mailenable: 0
> mailto: ""
> originalAddress: "01005__  "
> push: 0
> pushgroup: null
> pushpri: "0"
> pushsound: null
> telechat: null
> telegram: 0
> __proto__:
> $delete: ƒ (a,c,b)
> $get: ƒ (a,c,b)
> $post: ƒ (a,c,b)
> $query: ƒ (a,c,b)
> $remove: ƒ (a,c,b)
> $save: ƒ (a,c,b)
> toJSON: ƒ ()
> constructor: ƒ l(a)
> __proto__: Object
> headers:
> Accept: "application/json, text/plain, */*"
> Content-Type: "application/json;charset=utf-8"
> __proto__: Object
> jsonpCallbackParam: "callback"
> method: "POST"
> paramSerializer: ƒ (a)
> arguments: (...)
> caller: (...)
> length: 1
> name: ""
> prototype: {constructor: ƒ}
> __proto__: ƒ ()
> [[FunctionLocation]]: angular.js:11073
> [[Scopes]]: Scopes[2]
> transformRequest: Array(1)
> 0: ƒ (a)
> length: 1
> __proto__: Array(0)
> transformResponse: Array(1)
> 0: ƒ nc(a,b)
> length: 1
> __proto__: Array(0)
> url: "/api/capcodes/1610"
> __proto__: Object
> data:
> error:
> message: "console.debug is not a function"
> name: "TypeError"
> text: "TypeError: console.debug is not a function"
> __proto__: Object
> __proto__: Object
> headers: ƒ (d)
> status: 500
> statusText: ""
> __proto__:
> constructor: ƒ Object()
> hasOwnProperty: ƒ hasOwnProperty()
> isPrototypeOf: ƒ isPrototypeOf()
> propertyIsEnumerable: ƒ propertyIsEnumerable()
> toLocaleString: ƒ toLocaleString()
> toString: ƒ toString()
> valueOf: ƒ valueOf()
> __defineGetter__: ƒ __defineGetter__()
> __defineSetter__: ƒ __defineSetter__()
> __lookupGetter__: ƒ __lookupGetter__()
> __lookupSetter__: ƒ __lookupSetter__()
> get __proto__: ƒ __proto__()
> set __proto__: ƒ __proto__()

Any idea what the source could be of this error or how I can troubleshoot it further?

Thanks!

marshyonline, 2018-11-14:
Would you be able to send us the schema for you capcode and alias tables?

marshyonline, 2018-11-14:
Also, is there an error logged in the servers error log?

mitch2k, 2018-11-14:
Sorry, forgot to mention that. There are no errors in logs/node-app.stderr.log.

This is the schema:

```
CREATE TABLE capcodes ( id INTEGER PRIMARY KEY AUTOINCREMENT, address TEXT NOT NULL, alias TEXT NOT NULL, agency TEXT, icon TEXT, color TEXT, push INTEGER DEFAULT 0, pushpri TEXT, mailenable INTEGER DEFAULT 0, mailto TEXT, ignore INTEGER DEFAULT 0 , pushgroup TEXT, pushsound TEXT, telegram INTEGER DEFAULT 0, telechat TEXT);
CREATE TABLE messages ( id INTEGER UNIQUE, address TEXT NOT NULL, message TEXT NOT NULL, source TEXT NOT NULL, timestamp INTEGER, alias_id INTEGER, PRIMARY KEY(`id`), FOREIGN KEY(`alias_id`) REFERENCES capcodes(id) );
CREATE INDEX `msg_index` ON `messages` (`address`,`id` DESC);
CREATE INDEX `msg_alias` ON `messages` (`id` DESC, `alias_id`);
CREATE UNIQUE INDEX `cc_pk_idx` ON `capcodes` (`id`,`address` DESC);
CREATE VIRTUAL TABLE messages_search_index USING fts3(message, alias, agency);
CREATE TABLE 'messages_search_index_content'(docid INTEGER PRIMARY KEY, 'c0message', 'c1alias', 'c2agency');
CREATE TABLE 'messages_search_index_segments'(blockid INTEGER PRIMARY KEY, block BLOB);
CREATE TABLE 'messages_search_index_segdir'(level INTEGER,idx INTEGER,start_block INTEGER,leaves_end_block INTEGER,end_block INTEGER,root BLOB,PRIMARY KEY(level, idx));
CREATE TRIGGER messages_search_index_insert AFTER INSERT ON messages BEGIN
                    INSERT INTO messages_search_index (
                        rowid,
                        message,
                        alias,
                        agency
                    )
                    VALUES(
                        new.id,
                        new.message,
                        (SELECT alias FROM capcodes WHERE id = new.alias_id),
                        (SELECT agency FROM capcodes WHERE id = new.alias_id)
                    );
                    END;
CREATE TRIGGER messages_search_index_update AFTER UPDATE ON messages BEGIN
                    UPDATE messages_search_index SET
                        message = new.message,
                        alias = (SELECT alias FROM capcodes WHERE id = new.alias_id),
                        agency = (SELECT agency FROM capcodes WHERE id = new.alias_id)
                    WHERE rowid = old.id;
                    END;
CREATE TRIGGER messages_search_index_delete AFTER DELETE ON messages BEGIN
                    DELETE FROM messages_search_index WHERE rowid = old.id;
                    END;
```

davidmckenzie, 2018-11-14:
```
address: "01005__ "
```

davidmckenzie, 2018-11-14:
```
error:
message: "console.debug is not a function"
name: "TypeError"
text: "TypeError: console.debug is not a function"
```

davidmckenzie, 2018-11-14:
Turns out console.debug was only added in v8, I'll switch that back to console.log as workaround, 2 secs

davidmckenzie, 2018-11-14:
Done, try pulling fresh again now and see if that works, or at least produces more descriptive errors :)

DanrwAU, 2018-11-15:
>  "Error saving alias: [object Object]"

I've seen this before when i screw up the DB Structure while developing the query but it looks ok to me from what you've posted.

mitch2k, 2018-11-15:
Hi,

> Done, try pulling fresh again now and see if that works, or at least produces more descriptive errors :)

I updated to the latest version, and now it works. I can save again, and no errors in console. Thanks!

> ```
> address: "01005__ "
> ```
> Could be the space at the end of the address?

Hmm. I imported around 2400 capcodes from an excel file. It seems that some of them had space. However didn't had any issue (also with the previous install). 

What I do see is that when I do a map refresh now, the page keeps loading forever, but it seems that the refresh completes in 5-6 minutes.  During that time no new messages are being displayed.
In the console of the map refresh I see after a while:

> POST http://xxx:3000/api/capcodeRefresh net::ERR_EMPTY_RESPONSE
> (anonymous) @ angular.js:12587
> p @ angular.js:12332
> (anonymous) @ angular.js:12084
> (anonymous) @ angular.js:16832
> $digest @ angular.js:17971
> $apply @ angular.js:18269
> (anonymous) @ angular.js:26999
> dispatch @ jquery.min.js:3
> q.handle @ jquery.min.js:3
> 

and

> TypeError: Cannot read property 'error' of null
>     at admin.main.js:192
>     at angular.js:16832
>     at m.$digest (angular.js:17971)
>     at m.$apply (angular.js:18269)
>     at l (angular.js:12387)
>     at XMLHttpRequest.e (angular.js:12551) "Possibly unhandled rejection: {}"
> (anonymous) @ angular.js:14525
> (anonymous) @ angular.js:11008
> g @ angular.js:16858
> $digest @ angular.js:17971
> $apply @ angular.js:18269
> l @ angular.js:12387
> e @ angular.js:12551
> error (async)
> (anonymous) @ angular.js:12554
> p @ angular.js:12332
> (anonymous) @ angular.js:12084
> (anonymous) @ angular.js:16832
> $digest @ angular.js:17971
> $apply @ angular.js:18269
> (anonymous) @ angular.js:26999
> dispatch @ jquery.min.js:3
> q.handle @ jquery.min.js:3

DanrwAU, 2018-11-15:
What hardware are you running on?

The updating is very CPU intensive and takes around 50 seconds for me with 1700 Capcodes.

mitch2k, 2018-11-19:
Hi,

it's a VM with 1 vpcu (Xeon 2.1 ghz) and 2 gb RAM with SSD.

I tried again running the refresh directly on port 3000 to rule out haproxy issues. When starting the refresh the node /opt/pager proc goes to 99% cpu usage. After 3 minutes the ERR_EMPTY_RESPONSE is thrown in console. After 9 minutes the cpu usage drops to <1%. The page keeps on loading but after hitting F5 all is clear and the refresh is actually done as I see changes committed.

Additional phenomenon is that when I run the refresh when pagermon is behind cloudflare, I get a 504 after a minute because the page is not responding to the cloudflare proxy anymore.  

Nevertheless the refresh actually succeeds, but maybe it's a suggestion to avoid 'blocking' of the web gui by running the refresh asynchronously in the background (if that is even possible)?

PS: since this is not related to the original issue, let me know if you want me to create a separate issue.

davidmckenzie, 2018-11-19:
EC2 instance or a different VM? If EC2 instance it's probably running out of CPU burst credits, but sounds like it's not EC2 from your description.

How many messages in your DB?

Also, can you run the following in the sqlite db and paste the output? (e.g. `sqlite3 messages.db` from the console):

```
.schema
EXPLAIN QUERY PLAN UPDATE messages SET alias_id = (SELECT id FROM capcodes WHERE messages.address LIKE address ORDER BY REPLACE(address, '_', '%') DESC LIMIT 1);
```

First command will give you the full DB schema, we can see if anything odd in the indexes there. Second command will tell us whether the indexes are actually being used. Here's my output for the second one:

```
sqlite> EXPLAIN QUERY PLAN UPDATE messages SET alias_id = (SELECT id FROM capcodes WHERE messages.address LIKE address ORDER BY REPLACE(address, '_', '%') DESC LIMIT 1);
0|0|0|SCAN TABLE messages
0|0|0|EXECUTE CORRELATED SCALAR SUBQUERY 0
0|0|0|SCAN TABLE capcodes USING COVERING INDEX cc_pk_idx
0|0|0|USE TEMP B-TREE FOR ORDER BY
```

Lastly, if you wanna run these two it'll give us a more accurate time of the query:
```
.timer ON
UPDATE messages SET alias_id = (SELECT id FROM capcodes WHERE messages.address LIKE address ORDER BY REPLACE(address, '_', '%') DESC LIMIT 1);
```

Re non-blocking, I don't think it's possible at the moment - sqlite is single threaded by nature, so even if we were to break out the capcode refresh process into an async block, it would still block all the other sql commands. Definitely won't be a problem once we switch to a different db engine. :)

DanrwAU, 2018-11-21:
@DanrwAU Could this be done as a post alias save plugin? Have it update the database for only the address being updated? 
This should make it alot snappier and remove the need to do this manually?

davidmckenzie, 2018-11-21:
Problem is it needs to re-run for all aliases every time cos of the logic - if you ran it only for the alias you're saving, then you might overwrite more specific matches. E.g. if you have an alias for `88888888` and then add an alias for `8%`, the `8%` alias would match but the `88888888` should be the match.

closed 2019-01-13 by davidmckenzie