pagermon/pagermon #144: Final Message Validity Check

Joraeim, 2018-11-24:
In reader.js @ line 107, 

`  // filter out most false hits`
`  // if too much junk data, make sure '-p' option isn't enabled in multimon`
`  if (address.length > 4 && message) {`
`    var padAddress = padDigits(address,7);`
  
The `address.length > 4` was killing all of my POCSAG1200 messages.

I have four digit addresses on my system. I'll try and change it and submit a pull request, I just didn't want this to get lost before I got the pull request done.

Thanks for the awesome start!

DanrwAU, 2018-11-24:
I've changed this locally on my POCSAG 1200 instance to 2 without any issues.

davidmckenzie, 2018-11-26:
Tbh I made a lot of assumptions with that - I don't think I've ever seen a message come through (broken or not) with an address length < 4 on my system, so that check was purely based on observations of my local system.

Would be cool with change to 2 👌

Joraeim, 2018-11-26:
I think I also changed mine to two and it was working just fine. 
@davidmckenzie That's what I was thinking you'd probably done. As far as I can tell, POCSAG itself and multimon-ng don't limit the lower end of valid addresses. 
@marshyonline If you want to put in the PR, that's fine by me, otherwise I'll get around to it eventually. :-P
Thanks!

DanrwAU, 2018-11-27:
PR Done 
https://github.com/davidmckenzie/pagermon/pull/145

davidmckenzie, 2018-11-27:
Merged!

closed 2018-11-27 by davidmckenzie

marshyonline, 2018-11-27:
Happy Days!

Joraeim, 2018-11-27:
Awesome!