pagermon/pagermon #150: Errors when running $ pm2 start server/process.json

marto73, 2018-12-05:
Hi Guys,

I'm trying to run Pagermon on a fresh Ubuntu 18.04.1 LTS virtual machine. Sorry for the long post, but trying to include as much detail as possible.

The install went OK, but with a few errors, listed below.

$ npm install
	got errors, worked when run in the pagermon/server folder instead of the pagermon folder
	npm WARN deprecated node-uuid@1.4.8: Use uuid module instead
	npm WARN deprecated snekfetch@3.6.4: use node-fetch instead

	> sqlite3@4.0.4 install /home/andrew/pagermon/server/node_modules/sqlite3
	> node-pre-gyp install --fallback-to-build

	node-pre-gyp WARN Using request for node-pre-gyp https download 
	[sqlite3] Success: "/home/andrew/pagermon/server/node_modules/sqlite3/lib/binding/node-v57-linux-x64/node_sqlite3.node" is installed via remote
	pagermon@0.1.9-beta /home/andrew/pagermon/server

	(after the tree listing)

	npm WARN discord.js@11.4.2 requires a peer of bufferutil@^3.0.3 but none was installed.
	npm WARN discord.js@11.4.2 requires a peer of erlpack@discordapp/erlpack but none was installed.
	npm WARN discord.js@11.4.2 requires a peer of node-opus@^0.2.7 but none was installed.
	npm WARN discord.js@11.4.2 requires a peer of opusscript@^0.0.6 but none was installed.
	npm WARN discord.js@11.4.2 requires a peer of sodium@^2.0.3 but none was installed.
	npm WARN discord.js@11.4.2 requires a peer of libsodium-wrappers@^0.7.3 but none was installed.
	npm WARN discord.js@11.4.2 requires a peer of uws@^9.14.0 but none was installed.

When I type the command $ pm2 start server/process.json I have the following issues
	[PM2] Spawning PM2 daemon with pm2_home=/home/andrew/.pm2
	[PM2] PM2 Successfully daemonized
	[PM2] cron restart at 1 0 * * *
	[PM2][WARN] Applications pagermon not running, starting...
	[PM2][ERROR] script not found : /home/pagermon/server/app.js

after reading issue #28 I tried 
	typed $ pm2 monitor
	typed $ pm2 install server/app.js 

I'm having a similar issue to #28 in that the app.js is constantly restarting every 4 to 5 seconds but none of the solutions there have worked. The dump from PM2 
		Error: ENOENT: no such file or directory, open './config/config.json'
		    at Object.fs.openSync (fs.js:646:18)
		    at Object.fs.writeFileSync (fs.js:1299:33)
		    at Object.<anonymous> (/home/andrew/pagermon/server/app.js:45:8)
		    at Module._compile (module.js:652:30)
		    at Object.Module._extensions..js (module.js:663:10)
		    at Module.load (module.js:565:32)
		    at tryModuleLoad (module.js:505:12)
		    at Function.Module._load (module.js:497:3)
		    at Object.<anonymous> (/usr/local/lib/node_modules/pm2/lib/ProcessContainerFork.js:48:21)
		    at Module._compile (module.js:652:30)
	(

I have programming and Linux experience, but it's been over 15 years since I used it so I'm pretty rusty.
Any assistance you could provide would appreciated

DanrwAU, 2018-12-05:
Try copying default.json to config.json in the config directory. 
Looks to me like it's complaining it can't read that file. Can't say i've ever run into this issue on Ubuntu 18.04 and i break mine at least twice a week 😄 

NPM Install does need to be run from the server directory, i've been meaning to throw in a PR for the readme to fix that 😨 . 
The discord.js errors are normal, those are non-required dependencies

davidmckenzie, 2018-12-05:
It *should* autocreate the config file on first start. I might have buggered up the ordering at some point.

On the initial issue with running `pm2 start server/process.json` - that file needs to be modified to suit your environment - it's assuming the directory it's installed in is `/home/pagermon` vs the `/home/andrew/pagermon` you've got it in.

I suspect that's also why it's not finding the config - since you ran it from outside the server directory, the current working dir is one level too high. So if you either modify the process.json file to suit your environment, **or** just run the pm2 commands from within the server directory, either should help.

Note you may need to remove the current pm2 config first:
```
pm2 list
```
Then from the output, for each of the process names listed run:
```
pm2 stop <name>
pm2 delete <name>
```

Then should be good to go again with the `pm2 start server/process.json`

marto73, 2018-12-05:
Hi David,

Thanks for your quick response. Editing the process.json file was the first thing I did, but I forgot the user component, it's now /home/andrew/pagermon

I followed your other steps and it all looks good, pagermon is now running and I can access the website, but it's looking rather blank. Clinking any link after logging in changes the URL to the respective link but nothing else changes (as below)
[image]

davidmckenzie, 2018-12-05:
Looks like it's not loading any of the assets - some browsers can do that as a "feature" when the site doesn't have SSL. Not sure on the fix for that in Firefox, but guessing one of the buttons up in the address bar might let you live life dangerously and accept "external" assets that aren't HTTPS.

s3m1s0n1c, 2018-12-12:
@marto73 You still have issues or did u figure it out?

davidmckenzie, 2019-01-13:
Closing as inactive, feel free to reopen if any issues persist :)

closed 2019-01-13 by davidmckenzie