pagermon/pagermon #151: Bug with new date sorting

DanrwAU, 2018-12-06:
Getting this on the new version for new messages via the websocket 😭 

[image]

davidmckenzie, 2018-12-06:
This is spooky... it's getting modified at some point in the plugin handler...

If I drop some console logs here:

```
                            console.log(row);
                            logger.main.debug('afterMessage start');
                            pluginHandler.handle('message', 'after', row, function(response) {
                              console.log(row);
                              logger.main.debug(util.format('%o',response));
                              logger.main.debug('afterMessage done');
```

It's clearly changing the timestamp to datetime format after plugins have run:

```
  timestamp: 1544070748,
[..]
  timestamp: '2018-12-06 15:12:15',
```

davidmckenzie, 2018-12-06:
Ahahahahaha oh wow.

davidmckenzie, 2018-12-06:
```
logger.main.debug(data);
```

```
    format: format.combine(
[..]
        format.timestamp({format:"YYYY-MM-DD HH:MM:SS"}),
```

...

I got no words.

PR incoming.

closed 2018-12-06 by davidmckenzie

marshyonline, 2018-12-06:
We should probably recreate the release too @davidmckenzie :)

davidmckenzie, 2018-12-06:
Was thinking about that, then figured there's probably gonna be a couple more things sneak through that we didn't notice so will give it a couple of days :D

PR #152 (2018-12-06): In which is recounted the amusing way that don Quixote came to be dubbed a knight
Resolves #151

Winston decides that since he has been passed an object, clearly he has full rights to modify said object. Winston is incorrect.