pagermon/pagermon #156: How to get a Telegram user id ?

pyrog, 2018-12-10:
Several ways to get a user id:

- From your Telegram app:
  - Start a new chat with **userinfobot**
  - Start the bot `/start`
  - The bot give a human readable answer with your Id, First and Last name.

  or

  - Start a new chat with **jsondumpbot**
  - Start the bot `/start`
  - The bot give a machine readable answer in **json** format. Your id is `from.id` or `chat.id`

- From your web browser:
  - Open https://api.telegram.org/botXXX:YYY/getUpdates

DanrwAU, 2018-12-10:
@s3m1s0n1c was working on documentation for this as part of https://github.com/davidmckenzie/pagermon/issues/122 i think.

closed 2018-12-10 by pyrog

s3m1s0n1c, 2019-04-26:
This has been added to Wiki
https://github.com/pagermon/pagermon/wiki/Notifications#telegram-channelgroupuser-chat-id