pagermon/pagermon #157: Telegram: StatusCodeError:

pyrog, 2018-12-10:
Some `Telegram: StatusCodeError: `returned by Telegram in the log file:
- `400`: wrong user/chat id
- `403`: bot stopped
- `404`: wrong API key

pyrog, 2019-01-05:
Got status `400` when some characters are in the message:
```
message: 'Test 5b [',
message: 'Test 5f _',
message: 'Test 60 `',
```

pyrog, 2019-01-05:
Also, what is the behaviour with message containing double quotes?
Are they escaped?

See #180

davidmckenzie, 2019-01-14:
Just closing out support issues - if you still need any assistance feel free to jump on the new Discord server :) https://discord.gg/3VK7gSD

Alternatively - if you feel this is a bug in PagerMon (vs limitations of the Telegram API), please resubmit as a bug.

closed 2019-01-14 by davidmckenzie