pagermon/pagermon #158: Can't sign in

pyrog, 2018-12-10:
http://127.0.0.1:3000/login redirect to the login page without any error messages.
(With or without a good Username and Password)

This issue happened only with the localhost address.
See [passport/issues#482](https://github.com/jaredhanson/passport/issues/482#issuecomment-377126345)

marshyonline, 2018-12-16:
This may be an issue - I'm unable to login with no error being shown.
Expect i'm not logging into a local host - i'm hitting a host through Cloudflare

davidmckenzie, 2018-12-16:
Looks like you're using ipv6 there - or is that the cloudflare IP? If cloudflare's, then could be a trust-proxy setting since it's not showing your real IP. Whatever it's showing there would be stored as the session (I think)

If it is your IP, then could be something borked with ipv6, will have to dig into that a bit more.

marshyonline, 2018-12-16:
Yeah I'm using IPv6 - Cloudflare is forwarding the correct client IP's thankfully
This also happens when I bypass CF and hit the server direct over ipv4 and 6 :(

marshyonline, 2018-12-16:
Over IPv4(From my mobile teathered to a laptop) https://pastebin.com/JfVT6v8z
Pastebin beacuse i hate code blocks on github issues and it breaking nice formatting

marshyonline, 2018-12-16:
So far I have tried
- Clearing out the sessions DB
- Updating Passport to the latest version 

neither of these resolved the issue.

davidmckenzie, 2018-12-16:
How odd! Will see if I can replicate. Is your hostname set in env variables?

marshyonline, 2018-12-16:
removing the ENV and letting it sort itself has resolved it when access via hostname:3000
Still broken for it when accessing via the site URL tho :(

marshyonline, 2018-12-16:
I'll add to this that the setup was working for quite a while and I had no issues then one night logins failed.
Adding debugging to the login stuff shows its authing ok
when hostname  ENV is set to the site URL - it still fails to login 

Looks like it may be failing to save cookie?

DanrwAU, 2018-12-16:
I've been meaning to open an issue for something similar. 

I have 2 sites running 

root domain site.com
subdomain boringstuff.site.com 

Both have the same password set, i can login to the root domain, however logins to the subdomain fail unless the browser is in private mode, i suspect a cookie issue as well but haven't had a chance to look into it yet unfortunately.

marshyonline, 2018-12-16:
https://github.com/davidmckenzie/pagermon/blob/30c1367f98ddfec636fb825a7d620d42329d9c01/server/app.js#L114

This is what is causing the issue - Commenting this out allows me to log in regardless of url used

davidmckenzie, 2018-12-16:
Hmm that's what makes the cookie work across multiple domains (i.e. www and non-www). Wonder if a chrome update killed it.

marshyonline, 2018-12-16:
At first I thought it was a browser issue but I tried across multiple devices and browsers with no luck :(

timrowlands, 2018-12-19:
I am also having the same issue, however I am unable to log in to change the default user credentials.

timrowlands, 2018-12-19:
Sorry to ping again, as per:

> [pagermon/server/app.js](https://github.com/davidmckenzie/pagermon/blob/30c1367f98ddfec636fb825a7d620d42329d9c01/server/app.js#L114)
> 
> Line 114 in [30c1367](/davidmckenzie/pagermon/commit/30c1367f98ddfec636fb825a7d620d42329d9c01)
> 
>  sessSet.cookie.domain = '.'+hostname; 
> This is what is causing the issue - Commenting this out allows me to log in regardless of url used

Fixes the issue with logging in locally. I am not running in a proxied environment yet.

shutterfire, 2019-01-05:
Same error here - signing in at external devices does not work at all
Tried the actual ip adress, the hostname ("dietpi" in this case as well as "DietPi") and 0.0.0.0

Clicking on "Login" simply reloads the page without being logged in.

The fix mentioned here:

> [pagermon/server/app.js](https://github.com/davidmckenzie/pagermon/blob/30c1367f98ddfec636fb825a7d620d42329d9c01/server/app.js#L114)
>
> Line 114 in [...]

helped.

davidmckenzie, 2019-01-08:
I've just submitted PR #188 that should hopefully fix this - this is one of those "welp, worked for me" problems, so not sure the dealio

PR #188 (2019-01-08): make hostname thing optional
Should resolve #158 - doesn't even attempt to set the cookie domain unless explicitly true'd in a new env variable
closed 2019-01-08 by marshyonline