pagermon/pagermon #159: POP out Filter box for non admin's

s3m1s0n1c, 2018-12-11:
Hey Guys,

Is the Pop-out filter box meant to be for Non-Admin Users?

I think it looks very clunky if it is for Non-Admin users..

See Screeny below..

[image]

davidmckenzie, 2018-12-11:
Yeah bit useless at the moment... the idea there is that we can add more stuff to it later on. Like if we ever tackle #52 it could be useful to have a 'filter out' and a 'show matching' button. Or could have an 'open in new tab' button that gives you a nice pretty view of a description of the alias or something and list of the messages beneath.

s3m1s0n1c, 2018-12-11:
Is anyone really going use use Advanced Filtering?

I think keyword, Agency and Alias would safice..

Even so.. I dunno just seems clunky anyways even if advanced was there.. specially that 90% of users will prob be mobile users..

Just my 5 cents...

s3m1s0n1c, 2018-12-12:
Fair comments..

Will close this one then as its not really an issue and has been explained.

closed 2018-12-12 by s3m1s0n1c