pagermon/pagermon #162: Customize port number

112bzo, 2018-12-15:
Is it possible to adjust the port 3000 in a different port? because I have 2x Rpi running. One with Flex and one with POCSAG but wants to put this POCSAG on port 3001.

First running @ : http://86.86.74.22:3000/

112bzo, 2018-12-15:
Where do i find that file? i have PM2 running.

davidmckenzie, 2018-12-15:
Depends on your environment, but by default you'd place it in the server dir.

PM2 docos has more info on setting env variables: http://pm2.keymetrics.io/docs/usage/environment/

112bzo, 2018-12-18:
Sorry friends, can't find the file where i must chance the port setting. Like to have port 3001

marshyonline, 2018-12-18:
If you haven't moved the file, it's in the same folder as app.js

marshyonline, 2018-12-18:
You may also know it as this file if you didn't rename it
https://github.com/davidmckenzie/pagermon/blob/master/server/process-default.json

112bzo, 2018-12-19:
Case closed. I have made it. 

First Rpi [http://86.86.74.22:3000/](http://86.86.74.22:3000/) its the filter of my district
Second Rpi [http://86.86.74.22:3001](http://86.86.74.22:3001) its the Dutch network of pager alarms EMS

Thanks people.

closed 2018-12-23 by 112bzo