pagermon/pagermon #165: Duplicate filtering

ggreer123, 2018-12-27:
Is there a way to do the message filtering on just the message text? I can see that pagermon filters if multiple identically numbered and message content messages re received, but we have messages being sent (with identical content) to multiple pagers. I'm happy to hack the code if i can figure out where the filters currently live!

DanrwAU, 2018-12-27:
Not for duplicate messages on different addresses, No. 

There is an issue open for some way of folding them into 1 line  here https://github.com/davidmckenzie/pagermon/issues/30 but nothing has happened with it as yet.

davidmckenzie, 2018-12-27:
You could do some hackery on lines 513 and 515 of server/routes/api.js - just drop the `AND address` chunk and it'll filter on the message content only. Only problem is then you don't know who it was sent to, only the first alias/capcode that got it :)

ggreer123, 2018-12-27:
Sort of worked - fell over at midnight and came back with sql errors. I'm thinking that perhaps pagermon didn't know what to do when calculating times over midnight, but I'm only guessing. I'll wait for the real programmers to sort this one out!