pagermon/pagermon #166: Help it doens't work

JJS79, 2018-12-30:
Who can and wants to help me?

I got the Pagermon online but nothing comes in.

Where am I wrong?

I modified the file /client/config/config.json with the API-Key that I received on the website.

I also have a P2000 server running (the Netherlands) and there the messages come in so there is no error.
Even if I switch it off and restart Pagermon it does not work.

I have adjusted the RTL string for start decoding in the file /client/reader.sh

The string I use is:
rtl_fm -f 169.65M -M fm -s 22050 -d1 -p 67 | multimon-ng -a FLEX -t raw / dev / stdin

Personally, I think I see something small about the top, but I have no idea.

Thanks for helping

JJS79, 2018-12-30:
Hi DanrwAU,

Thanks for the info, I did not notice the addition "| node reader.js".

It works!

However, where can I find the log file of the cleint?
I can find that from the server.

Thanks again!

davidmckenzie, 2018-12-30:
Client side just logs to STDOUT - if you wanted that side logged to files you'd just have to `> pagermon.log` or something. :)

Docker isn't a requirement. There's a sample script in client/import.js you can use for importing capcodes from a PDW formatted file - if you have your capcodes stored in another format you can modify this script, or you could enter them manually in the UI.