pagermon/pagermon #167: Group id's

112bzo, 2018-12-30:
We are working on processing group notifications in pagermon. What we can not find is to remove the lines between these lines.

Or is it possible to process group notifications in another way?

[image]

davidmckenzie, 2018-12-30:
Can't really see what you mean there - could you post a screenshot instead, is this code in a fork somewhere or just on your local copy?

112bzo, 2018-12-30:
Now the Pagemon works like this:

- Date - Time - Capcode - tekst
- Date - Time - Capcode - tekst
- Date - Time - Capcode - tekst

But we like to have a singel call into one line (like my screen)

- Date - Time - Capcode - tekst
                      - Capcode  - tekst
                       - capcode - tekst

And that whithout line's between.

davidmckenzie, 2018-12-30:
Yeah would have to see the code you're working from - send through a pull request and I can take a look :)

DanrwAU, 2018-12-31:
I'm a little confused, what is the end result your looking for with this one? 😄

woestric, 2019-01-03:
Hi. Maybe I can elaborate on this as 112bzo was asking on my behalf. The emergency messages in the netherlands are mostly sent to multiple capcodes. So we like to group the messages. One message line and underneath all the capcodes with aliases it was sent to. We get a better oversight with this.
I have managed to adjust the queries to determine if the line is one of a group. 
I seems not possible to combine the messages and capcodes in one table row because database row to table row is one on one.
So now we are looking at changing the border box so that the boxlines between rows with only capcodes is not visible. That seems not possible because of the use of an external template.
I have still not succeeded in overriding the external css in this matter.
Main question is: how to override the external border box around the rows.
@davidmckenzie Not ready to share my code yet, Affraid it is to messy ;-)
It is a local copy and thus far I have change api.js, index.ejs, menu.ejs and style.css to get the following result.
[image]

davidmckenzie, 2019-01-03:
Yeah without seeing the code I won't be much help ;) when I was looking at this, my thought was to have an expander on the left of combined rows - click the button and it would just display a span within the row, with linebreaks instead of adding additional rows... does that make sense?

Other way you could do it - use rowspan on the other columns. Not sure what this would look like but could be worth a go.

davidmckenzie, 2019-01-03:
Hmm yeah this is a fun one. Just been thinking through how best to solve this in the back end - ideal world we'd send an array of alias matches as part of each message object. Easy enough to do with some mangling of the object, but it messes with pagination since we're limiting the DB query itself - if you had 10 rows showing per page, and there were 2 messages sent to 5 recipients, the user would only be displayed 2 unexpanded rows, which totally defeats the purpose of mushing them together.

Now looking at whether we can do something with group_concat:
```sql
sqlite> SELECT messages.message, group_concat(messages.address) FROM messages GROUP BY messages.message
   ...> ORDER BY messages.timestamp DESC LIMIT 10 OFFSET 0 ;
message|group_concat(messages.address)
Test message, disregard|1000000,1000000,1000011,1000000,1000000,1000000,1000000,1000000,1000000,1000000,1000000,1000000,1000000,1000004,0000004
Thu  6 Dec 2018 15:40:48 AEDT Test message, disregard|1000000
Thu  6 Dec 2018 15:37:19 AEDT Test message, disregard|1000000
Thu  6 Dec 2018 15:32:27 AEDT Test message, disregard|1000000
Thu  6 Dec 2018 15:32:00 AEDT Test message, disregard|1000000
Thu  6 Dec 2018 15:28:06 AEDT Test message, disregard|1000000
Thu  6 Dec 2018 15:27:32 AEDT Test message, disregard|1000000
Thu  6 Dec 2018 15:24:28 AEDT Test message, disregard|1000000
Thu  6 Dec 2018 15:23:04 AEDT Test message, disregard|1000000
Thu  6 Dec 2018 15:22:58 AEDT Test message, disregard|1000000
```

Only trouble is that list of addresses is going through the full DB... need to find a way to concat on neighbouring rows only or something.

davidmckenzie, 2019-01-04:
Actually... in the full text search message get we've been grabbing unrestrained batch of matches then just handling pagination after the query without any major performance impacts, may be worth doing the same on the main message get.

@DanrwAU @marshyonline I think you guys have the biggest DBs at the moment... what's the runtime like on this SQL?

```
SELECT messages.*, capcodes.alias, capcodes.agency, capcodes.icon, capcodes.color, capcodes.ignore, capcodes.id AS aliasMatch 
 FROM messages
 LEFT JOIN capcodes ON capcodes.id = messages.alias_id WHERE capcodes.ignore = 0 OR capcodes.ignore IS NULL
 ORDER BY messages.timestamp;
```

For me:

```
sqlite> .timer on
sqlite> SELECT messages.*, capcodes.alias, capcodes.agency, capcodes.icon, capcodes.color, capcodes.ignore, capcodes.id AS aliasMatch
   ...>  FROM messages
   ...>  LEFT JOIN capcodes ON capcodes.id = messages.alias_id WHERE capcodes.ignore = 0 OR capcodes.ignore IS NULL
   ...>  ORDER BY messages.timestamp;
<snip>
Run Time: real 0.331 user 0.105426 sys 0.060923
```

vs with limit 10:

`Run Time: real 0.050 user 0.026682 sys 0.010572`

DanrwAU, 2019-01-04:
DB1:
`Run Time: real 5.890 user 0.713131 sys 1.014042 `

```
 sqlite> SELECT count(*) FROM messages;
217931
```
DB2:
`Run Time: real 4.478 user 0.699169 sys 0.960878 `
```
sqlite> SELECT count(*) FROM messages;
173378
```

davidmckenzie, 2019-01-04:
Lame, definitely too long for every page load :(

DanrwAU, 2019-01-04:
My Page loads are already not the greatest 😛 

```
1|pagermon  | 2019-01-04 11:42 +11:00: init: 46.421ms
1|pagermon  | 2019-01-04 11:42 +11:00: sql: 646.320ms
1|pagermon  | 2019-01-04 11:42 +11:00: send: 1.544ms
```

davidmckenzie, 2019-01-04:
OOF... I really need to get around to #10 🤣

DanrwAU, 2019-01-04:
I actually oepned VSCode at 8am to work on the Knex stuff, i think i've done 1 line so far .... lol

davidmckenzie, 2019-01-04:
Still working on the styling, but looking at something like this:

[image]

The real shite part is it uses nested tables, which are gross. The columns in them are never going to line up properly (using rowspan was a lost cause due to the ng-repeat), so I'm thinking it's probably easiest to style them differently - drop the colour background on the capcode and maybe have a play with alignments to get it to a point where it's not so jarring to have the misaligned columns.

I'll also make it so that clicking the + expands/contracts the list, with a button next to the notifications enable/disable button to have them expanded by default.

davidmckenzie, 2019-01-04:
Also, I *really* hate the duplication of code in index.ejs at the moment... I'd like to pop it out to a macro possibly.

PR is here for visibility: https://github.com/davidmckenzie/pagermon/pull/171

davidmckenzie, 2019-01-04:
What do you guys think of this? Kinda distracts from the misalignment

[image]

davidmckenzie, 2019-01-04:
Close to something workable now - only blocker is the bloody colspans - since half of those columns are hidden on mobile it's spanning across the message column when on mobile. Will need to see if I can set the columns to 0 width instead of just `display: none`'ing them.

woestric, 2019-01-04:
Love how you guys taking this up. Personally I would not prefer to have to click a button. 
I was also thinking about nested tables but did not want to go that way either.
Here is the query I know use to get the previous and following message from the database:
`SELECT  m1.*, m2.message as lastmessage, m2.alias_id as lastalias, m3.message as nextmessage, m3.alias_id as nextalias, capcodes.alias, capcodes.agency, capcodes.icon, capcodes.color, capcodes.ignore, capcodes.id AS aliasMatch, 'false' as comparebef, 'false' as comparenext FROM messages m1 
       INNER JOIN messages m2 ON m2.id = (m1.id+1)
       INNER JOIN messages m3 ON m3.id = (m1.id-1)
      INNER JOIN capcodes ON capcodes.id = m1.alias_id WHERE capcodes.ignore = 0
       ORDER BY m1.id DESC,m1.timestamp DESC ;`

takenalias, 2019-01-04:
I've gotta agree, where possible it would be better not to click a button.
But if it can't be done any other way, what ever works.

DanrwAU, 2019-01-04:
Looking at the code, it looks like @davidmckenzie  has made this save in the cookie, so it's configurable whether it is expanded or not by default. 

Correct me if i'm wrong 😉

woestric, 2019-01-04:
I am quite fine with the layout right now if only I could get the lines between messages and not detaik capcode rows.
But there are a few more problems still ;-)
Thus far I didn't want to change your db definitions because that is to invasive. Therefore I looked on the query side.
RIght now I have to problems left:
1 - How to handle if one of the capcodes is not in my alias db. I think I have this one covered with adding the last and next alias and at this to the compare:
`  
                if (result.lastmessage == result.message && !!result.lastalias) {
                 result.comparebef = 'true';
                } 
                if (result.nextmessage == result.message && !!result.nextalias) {
                 result.comparenext = 'true' ;
                }`
2 - Somtimes when it's busy, different messages are sent mixed so:
message 1 capcode 1
message 1 capcode 2
message 2 capcode 1
message 1 capcode 3 etc

As message 2 is not in my area of interest, I don't need to see it but it messes up the query.
This one is harder to solve and I think I have to limit the filling of the table with only records with aliasmatch. Kind of what you do with ignore on.

BTW. In my opnion all of this is only valid in PDW mode the way we use it in the netherlands with P2000 messages.

davidmckenzie, 2019-01-04:
> Looking at the code, it looks like @davidmckenzie has made this save in the cookie, so it's configurable whether it is expanded or not by default.

Yep - I'm actually thinking of hiding the big expander toggle span when the cookie is enabled, looks much neater that way, and would be much closer to what @woestric et al are looking for. :)

> 1 - How to handle if one of the capcodes is not in my alias db. I think I have this one covered with adding the last and next alias and at this to the compare:
> 2 - Somtimes when it's busy, different messages are sent mixed so:

Yeah those caveats make it difficult. Second one could probably be handled if the merging is being done outside of the SQL query - just test within a certain time range.

With the SQL you posted - is that only bringing in the neighbouring messages (so 3 max)? Or is it looping to run until it finds no more neighbouring matches? May be worth benchmarking the two to see which runs better on the big DBs - if it's greatly increasing the number of queries run on each request then may see very poor performance as DB size increases. One of the annoying problems with SQLite, especially when it doesn't handle concurrency at all.

woestric, 2019-01-04:
No. I didn't want to loop because affraid of performance problems. So with getting messages it is just as I gave you. Before and after.  Kind of tricky because of de ordering desc before is actually +1 and after -1
Just corrected an error in the post query. There I only look for -1 off course. Accidently I named it next in stead of after.
Now testing if pushing also workes correctly.
I am just getting to know the angularis stuff combined with the tags. I still think fewer ng-if shoud work also.

But I really love your application by the way, great work!

112bzo, 2019-01-04:
> What do you guys think of this? Kinda distracts from the misalignment
> 
> [image]
> 
> This is it collapsed:
> 
> [image]

i like this one.....

woestric, 2019-01-07:
HI. Looks OK. I'd prefer to have the the colour background on the button removed. I think it's getting to 'colourful'  ;-)

woestric, 2019-01-07:
Crashed my sd/pagermon instance without proper backup/image. So I have to remodel my efforts.
But noticed that, even we get this query page as we want, we still have a major problem with pushing.
Because that happens per message. 
Shoud be nice if the push messages waits until all groupsmessages are in. Seems pretty hard to me but I am thinking in looking into that. 
Other option could be that, after last groups message pushed automatically an update page command runs. Is that even possible?

davidmckenzie, 2019-01-13:
> I'd prefer to have the the colour background on the button removed. I think it's getting to 'colourful'

Yeah I kinda agree, feels like too big a block of high contrast colour.

> But noticed that, even we get this query page as we want, we still have a major problem with pushing.
Because that happens per message.
Shoud be nice if the push messages waits until all groupsmessages are in. Seems pretty hard to me but I am thinking in looking into that.
Other option could be that, after last groups message pushed automatically an update page command runs. Is that even possible?

Should be do-able but does need some thinking - I figure when we push we can just test content and add to previous message's `allAddresses` array if it's a group, then refresh scope and it *should* just magically work.

At any rate, I'm gonna be putting this one on the backlog for a while as #174 takes precedence.