pagermon/pagermon #168: Docker issue

Neoakb, 2018-12-31:
I installed using the docker method.

I can see pagermon on localhost:3000, but when I try to login using the admin/changeme combo it does nothing.

Not sure what to do from here. Thanks for your help.

DanrwAU, 2018-12-31:
This is possibly the same as https://github.com/davidmckenzie/pagermon/issues/158 

Try the fix of commenting out the line given by @marshyonline  in that issue

Neoakb, 2018-12-31:
That worked, thank you.

Sorry to bother, but what do I do next? I see I need a database pointing to pagermon?

DanrwAU, 2018-12-31:
You will need a client receiver to send messages to pagermon. 

Usually used are Multimon-NG for Linux (RaspberryPi etc) or PDW for Windows

Neoakb, 2018-12-31:
I'd be using multimon which I have installed, but I don't know what command I need. Again sorry for the noob questions.

DanrwAU, 2018-12-31:
Take a peek at reader.sh and reader.js in the client folder.
 i also helped someone with this yesterday which may assisthttps://github.com/davidmckenzie/pagermon/issues/166

Neoakb, 2018-12-31:
We're getting there here's the output ` rtl_fm -f 169.65M -M fm -s 22050 -d1 -p 67 | multimon-ng -a FLEX -t raw /dev/stdin | node reader.js
Found 1 device(s):
multimon-ng 1.1.6
  (C) 1996/1997 by Tom Sailer HB9JNX/AE4WA
  (C) 2012-2018 by Elias Oenal
Available demodulators: POCSAG512 POCSAG1200 POCSAG2400 FLEX EAS UFSK1200 CLIPFSK FMSFSK AFSK1200 AFSK2400 AFSK2400_2 AFSK2400_3 HAPN4800 FSK9600 DTMF ZVEI1 ZVEI2 ZVEI3 DZVEI PZVEI EEA EIA CCIR MORSE_CW DUMPCSV X10 SCOPE
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Tuner gain set to automatic.
Tuner error set to 67 ppm.
Tuned to 169903575 Hz.
Oversampling input by: 46x.
Oversampling output by: 1x.
Buffer size: 8.08ms
Exact sample rate is: 1014300.020041 Hz
Allocating 15 zero-copy buffers
Sampling at 1014300 S/s.
Output at 22050 Hz.
module.js:549
    throw err;
    ^

Error: Cannot find module 'nconf'
    at Function.Module._resolveFilename (module.js:547:15)
    at Function.Module._load (module.js:474:25)
    at Module.require (module.js:596:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/home/neonfire/pagermon-master/client/reader.js:25:13)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
`

DanrwAU, 2018-12-31:
run from the client folder:
`npm install`

Neoakb, 2018-12-31:
Got that.

here's the output:
rtl_fm -f 169.65M -M fm -s 22050 -d1 -p 67 | multimon-ng -a FLEX -t raw /dev/stdin | node reader.js
Found 1 device(s):
multimon-ng 1.1.6
  (C) 1996/1997 by Tom Sailer HB9JNX/AE4WA
  (C) 2012-2018 by Elias Oenal
Available demodulators: POCSAG512 POCSAG1200 POCSAG2400 FLEX EAS UFSK1200 CLIPFSK FMSFSK AFSK1200 AFSK2400 AFSK2400_2 AFSK2400_3 HAPN4800 FSK9600 DTMF ZVEI1 ZVEI2 ZVEI3 DZVEI PZVEI EEA EIA CCIR MORSE_CW DUMPCSV X10 SCOPE
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Tuner gain set to automatic.
Tuner error set to 67 ppm.
Tuned to 169903575 Hz.
Oversampling input by: 46x.
Oversampling output by: 1x.
Buffer size: 8.08ms
Exact sample rate is: 1014300.020041 Hz
Allocating 15 zero-copy buffers
Sampling at 1014300 S/s.
Output at 22050 Hz.

but when I run :
$ ./reader.sh
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

No matching devices found.
Input died!

DanrwAU, 2018-12-31:
That usually happens when one step in the process is failing. 

I suggest taking off the` | node reader.js ` from the end of the command and confirming rtl_fm and multimon are actually processing messages correctly.

Neoakb, 2018-12-31:
rtl_fm -f 169.65M -M fm -s 22050 -d1 -p 67 | multimon-ng -a FLEX -t raw /dev/stdin 
multimon-ng 1.1.6
  (C) 1996/1997 by Tom Sailer HB9JNX/AE4WA
  (C) 2012-2018 by Elias Oenal
Available demodulators: POCSAG512 POCSAG1200 POCSAG2400 FLEX EAS UFSK1200 CLIPFSK FMSFSK AFSK1200 AFSK2400 AFSK2400_2 AFSK2400_3 HAPN4800 FSK9600 DTMF ZVEI1 ZVEI2 ZVEI3 DZVEI PZVEI EEA EIA CCIR MORSE_CW DUMPCSV X10 SCOPE
Enabled demodulators: FLEX
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Tuner gain set to automatic.
Tuner error set to 67 ppm.
Tuned to 169903575 Hz.
Oversampling input by: 46x.
Oversampling output by: 1x.
Buffer size: 8.08ms
Exact sample rate is: 1014300.020041 Hz
Allocating 15 zero-copy buffers
Sampling at 1014300 S/s.
Output at 22050 Hz.

Seems like multimon can find my rtlsdr, but reader.js can't?

DanrwAU, 2018-12-31:
AHA Yes, i have had this problem.  You need to put the full path to rtl_fm into reader.sh OR install rtl_fm using apt-get (if memory serves me correctly)

davidmckenzie, 2018-12-31:
Those pyenv errors look like permissions on your home dir might have changed, not sure whether that would be related though.

The `reader.sh` is just a super simple example bash script - have you modified it to match the rtl_fm command that you normally run? Of particular note would be dropping the `-d 0101` bit, as this selectively chooses a device (which doesn't exist on your box).

Neoakb, 2018-12-31:
That did it!, had to also change to FLEX and a local Flex freq, but that did it!

(any ideas on fixing the home dir permissions? I totally get if it's not your problem, but you seem to know more than me lol)

davidmckenzie, 2019-01-01:
`sudo chown -R neonfire /home/neonfire` should do it for the home dir :)

The garbled messages usually just mean a bit of tuning needed of your rtl_fm or multimon arguments, dropping or raising gain can help - basically just something where you'd mess around until you find the right settings

Neoakb, 2019-01-01:
That did it as far as the pyenv problems! I'll work on my settings for the other stuff and see what happens. Thanks!

Neoakb, 2019-01-02:
So I tested my other command (up a couple comments) and remembered I needed gqrx to pipe the audio. so yes, it works fine. I added node reader.js and ran it from /pagermon/client and it loaded fine, but stops after a few mins. Well it just doesn't post anything new, it doesn't crash. 

I think the problem is in reader.js